MoonModules/WLED-MM #1: Include ledmaps for 2D

dosipod, 2022-11-08:
Hi , 
Some use custom maps a lot as a practical way of  customizing the effects on the fly  for custom fixtures like cubes , it is currently working fine  with WLED 0.14.0-mdev (build 2210201) when the matrix is set to 1D   but not when you setup the unit with 2D . In that case ledmaps are ignored . 

**Is your feature request related to a problem? Please describe.**
Low priority feature request as it seems little user base 

**Describe the solution you'd like**
Ledmaps are available with 2D 

**Describe alternatives you've considered**
na

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

MoonModules, 2022-11-09:
You say it is working fine with WLED 0.14.0-mdev, I tried this:

[image]

but got this:

[image]

Is this also the way you do it?

MoonModules, 2022-11-09:
And this also not working for me: 

[image]

it can show the ledmap which has been set in the api, but changing it does nothing (old values still in the api), does it work for you?

dosipod, 2022-11-09:
Yes ledmaps  works fine with 14.0-mdev (build 2210201)  when in 1D mode . But  adding "ledmap":1 manually or selecting from list do not have any effect with 2D  setup . 
We never go any error in our case when adding that to a preset , it is just ignored  .Might be you see some other errors . We have tested the same on two units with 16x16 and 8x24 and we seen the same behavior 
[image]

ewoudwijma, 2022-11-15:
And working on ledmaps step by step: first fixed some stuff in WLED MM and added a pull request to WLED AC: https://github.com/Aircoookie/WLED/pull/2885.

Also for me to learn ledmaps ;-)

dosipod, 2022-11-15:
Sorry to copy and paste  but we sent that in discord , We are testing 0.14.0.3.0 ,in 1D the ledmaps are picked up and will apply correctly to a preset.
However selecting any other effects will reset to no ledmap and you have to save a preset for each effect with the ledmap   . Also still as before ledmaps are ignored in 2D

ewoudwijma, 2022-11-19:
I left some wrong code. Just removed it in latest commit.

Can you test and let me know if okay, then I can close this issue

dosipod, 2022-11-20:
Using  0.14.0.4.2 (build 2211192) we can selecting any other effects will no longer reset the ledmap so that part is solved .
For including 2D in ledmap , you may take your time  as there is no rush , would be nice if you have a discussion open for that on gh as this  post was meant as  an enhancement to include 2D  and might tell us if there is much interest or not

ewoudwijma, 2022-11-20:
I personally think that it is core functionality and not logical to have it not working in 2D and think it is doable to take the segment to logical matrix mapping, 2D uses, and apply ledmap to it:

[2,1,0] and [1,0,2] = [0->2->0, 1->1->0, 2->0->1] = [0,0,1]

Does that make sense?

dosipod, 2022-11-21:
Yeah agreed it is core functionality , one area is confusing to us is that 2D mapping is already used 
with  custom effect and hence we though that its already supported  . I am not really sure of the outcome 
with different fixture but we are hoping existing 2D effects would look better when correctly mapped.  

We just sound out that gaps in ledmaps is possible with -1 which is an example of undocumented feature
which I hope would help in tweaking 2D effects  without the need to code new ones  for none standard fixtures.

ewoudwijma, 2022-11-23:
How is it used in custom effects then? It uses setPixelColor so it would go through the same logic …

ewoudwijma, 2022-11-28:
@dosipod Is this change working as it should and can we close this issue and submit a pullrequest to add this to upstream?

dosipod, 2022-12-01:
@ewoudwijma  
Hello , Yes ledmaps  are now available  with 2D  setup as requested . Attached is the small demo we did and you tested  .  You may close or keep for others to verify .
Due to exams period we were not able to  customize  2D effects  using ledmaps  to have **good  looking** outcome for none standard fixtures  such as  cubes  which is the real goal but that could be another topic along with rotated panels ,3d and few other items that we could discuss in another thread . Thank you 
[image]

ewoudwijma, 2022-12-16:
Closed as it is implemented in WLED MM

closed 2022-12-16 by ewoudwijma