MoonModules/WLED-MM #2: No longer able to save presets after factory reset

dosipod, 2022-11-16:
### What happened?

Very strange issue I hope someone could confirm  , when you flash to 0.14.0.3.0 (build 2211152) with OTA  presets will work but if you do a factory reset  then you no will longer be able to save any presets unless you uncheck use current state . Also if you serial  flash  0.14.0.3.0 then you will not be able to save presets same as above and this  was observed with multiple units .   I am not sure if that issue was there in earlier versions and we missed that as last one we tested was 0.14.0.1.0
[image]

### To Reproduce Bug

factory reset  and try to save presets 

### Expected Behavior

presets are saved 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0.3.0 (build 2211152)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-11-16:
Could be related to https://github.com/Aircoookie/WLED/issues/2886

dosipod, 2022-11-20:
Tested in MM 0.14.0.4.2 (build 2211192)  and we can save presets after factory reset , you may close as you see fit

closed 2022-11-20 by ewoudwijma