MoonModules/WLED-MM #6: Setup https://moonmodules.github.io/WLED-Docs/

ewoudwijma, 2022-11-28:
Created this issue to discuss how we should setup MoonModules documentation using https://moonmodules.github.io/WLED-Docs/ 

This is a fork of https://kno.wled.ge and is in initial setup.
- perfect symbiose of upstream info + fork info IMO.
- supported by GitHub workflows so any commit towards it will automatically update the pages.

See also https://github.com/atuline/WLED/issues/235

ewoudwijma, 2022-11-28:
Current guidelines

- Welcome contains an extra paragraph on top called Welcome to WLED MoonModules, rest of the page is a 1:1 copy of upstream
- Basics, Features, Interfaces, Advanced, About is a 1:1 copy of upstream
- Added MoonModules, 2D and Sound Reactive
- MoonModules: features in WLEDMM not (yet) in upstream (except 2D and SR)
- 2D: 2D info
- Sound Reactive: SR info
- Some pages have been copy pasted from WLEDSR wiki, unmodified (e.g. Fork Comparison, 2D settings, SR settings, Digital Microphone Hookup), they need to be changed
- Every pages may contain a section at the bottom called Fork specific info (or similar) where differences compared to MM are described

Any comments?

dosipod, 2022-12-12:
Looks okay ,just small  comments , I do not think normal users will be interested to use MM much so you can skip the basic info that is already in the wled KB and focus on the more advanced functions  . Also as an example  I can not find reference to jmap in search

ewoudwijma, 2022-12-13:
>skip the basic info that is already in the wled KB
* This is just a GitHub repo and a fork of kno.wled.ge so we get this info by default and will kept up to date if synced with upstream
* Unless decided otherwise, this work will not edit existing WLED AC pages but only add in the new Tabs
* The big advantage of this being a fork we have all info in one place

>as an example I can not find reference to jmap in search
It's still WIP and I just added the map text here: https://moonmodules.github.io/WLED-Docs/moonmodules/mappings/ so now you can search on it, not a lot if info though  ;-)

ewoudwijma, 2022-12-13:
To make the next step in improving this repo, I made a 1:1 copy of everything what is on the WLEDSR / Wiki page (and removed stuff from the 2D and Sound Reactive tabs).

[image]

As a 'test' I added info similar as in WLED AC wiki on some of the current WLED SR Wiki pages as follows:

[image]

I am not sure yet what to do with this. Should I add this to all pages?
@softhack007 what do you think as you are the main editor, what are the next steps here?

softhack007, 2022-12-13:
> I am not sure yet what to do with this. Should I add this to all pages?
> @softhack007 what do you think as you are the main editor, what are the next steps for the WLEDSR wiki?

Hi, currently I don't update the SR Wiki so often, but a few times already added some details after we had discussion on discord or on gh bugreports.

I think we should keep the wiki pages  where they are, and add something like you suggested - like "latest/updated info is available in wled-docs". Because some people learned how to find them.

Seems I need to learn the new system ;-) Generally its ok for me to make new edits in wled-docs only.

Another small thing - we need to adjust the "help" links in SRWLED to point to the new locations.

ewoudwijma, 2022-12-13:
Indeed maybe better 'leave it as is' and also keep the links in the app the same. Think it is better after some more thoughts.

Maybe our update procedure can be driven by questions on discord or on just what we think needs to be done:

1) question requires update / clarification in documentation
2) if SR impact then update WLEDSR/Wiki documentation
3) if 0.14 / MM impact then migrate info in WLEDMM/Docs from WLEDSR tab to the right place, so we slowly build our new documentation (MM specific to MoonModules tab and 0.14 general to 2D/SoundReactive tab)

closed 2023-03-28 by MoonModules