MoonModules/WLED-MM #8: Unable to set segment name

simonemarin, 2022-12-10:
### What happened?

I am not able to set a segment name anymore.

### To Reproduce Bug

Clicking in the pencil icon and setting the text field, then using the check mark (I guess that is like the APPLY button in 0.13.3 ?), but the name is not applied

### Expected Behavior

In 0.13.3 you can set the name of  a segment, there is an apply button that is not present anymore to apply the changes.

### Install Method

From srg74 firmware repository

### What version/release of MM WLED?

(WLEDMM_0.14.0.10 esp32_16MB_max.bin) build 2212060

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I installed this version to get the scrolling text function going, that's why I want to change the segment name.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

simonemarin, 2022-12-11:
I noticed that this only happens in 2D mode, if I set the strip to 1D mode in settings then I can change the segments' names.

simonemarin, 2022-12-11:
Following in https://github.com/Aircoookie/WLED/issues/2945

ewoudwijma, 2022-12-16:
Closed as upstream fixed this and upstream fix has been synced with WLED MM