MoonModules/WLED-MM #9: Prevent effects from changing the defaults for Segments or Palette

dosipod, 2022-12-16:
We have noticed  that some of the effects do not default to pixels (in  expand 1D FX ) in segment setting even when manually changed   .Also some of the effects will change the default  Palette   , We think  an effect should not change the segment setting or Palette but here it will change which will apply to the next effect selected .

Some examples of effects that will change segment options : ( fixable by m12=0 for 1DFX part  ) 
Android ( default to Bar ) 
Bouncing balls ( default to Bar ) 
DJ Lights ( default to Arc )
Drip ( default to Bar )
Fire 2012 ( default to Bar )
Flow ( default to Bar )
Freqmatrix ( default to corner )

Also an example of an effect that will change Palette  is   Aurora effect  ( we just got blaz answer in discord but looking for my dictionary to translate to English as I do not get how to fix that ) 

This is tested on MM 0.14.0.10 (build 2212060)  . Sorry for the messy post  as this is just a reminder for what was discussed in discord . Thanks a lot for the help as this fork is keeping us busy with new features that we really like as it seems cutting edge

ewoudwijma, 2022-12-16:
Please move this issue to WLED AC as this WLED AC functionality (MM just gets this from syncing with that)

dosipod, 2022-12-16:
Sure , I thought expand 1D FX  is only in MM but I guess not

MoonModules, 2023-03-28:
Hey @dosipod,

Think this was discussed with upstream therefore closing here

closed 2023-03-28 by MoonModules