MoonModules/WLED-MM #12: DJ Light: not as good as in SR 0.13.3

softhack007, 2023-01-19:
### What happened?

From Artacus @ discord — today 17:03
On MM fork, I’d always get green dot at top left pixel.  On AC, it was barely reacting to sound and all the bars in a given frame would be the same color. So while it looked like it it was working, when it was side by side with a 0.13 render, it looked terrible in comparison.

### To Reproduce Bug

see above

This is a "reminder to myself" -> need to review the source code and compare it to SR 0.13.3 code

### Expected Behavior

looks as awesome as always :-)

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

WLED main development branch

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MoonModules, 2023-03-28:
@softhack007 should we close this?

softhack007, 2023-03-29:
> were you aiming to match the original output out just to give an "update" to make look pretty again?

Both actually.
First i tried to just restore the original look, by changing the frequency band according to the new mapping in 0.14.
That still gave me too much white and pastels, so I've tried to go RGB->HSV, then change saturation, then back to RGB. This looks better, but still could need some tweaking of parameter.

Finally I recognized the original code worked because the frequencies in 0.13 were not "equalized" as they are in 0.14. So I tried some other color mapping. That's still experimental, as I recognized idon't know enough about how the original effect was expected to look.

softhack007, 2023-04-01:
looks better with 11a02bb81463b496abc82aaa8c8e49ecca8c110c .
Ticket closed - further enhancements and tuning will be "normal work".

closed 2023-04-01 by softhack007