MoonModules/WLED-MM #13: 2D can't be disabled if the custom effect mod is enabled

Fonta, 2023-01-19:
### What happened?

If you use -D DISABLE_2D in the build flags, it will fail compiiling the custom effect usermod.

[image]

### To Reproduce Bug

Add -D WLED_DISABLE_2D to common_mm -> build_flags_max.

[image]

### Expected Behavior

Succesful build when you disable 2d

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

Latest commit

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Compiling .pio/build/esp32_4MB_max/src/usermod.cpp.o
Compiling .pio/build/esp32_4MB_max/src/usermods_list.cpp.o
Compiling .pio/build/esp32_4MB_max/src/util.cpp.o
Compiling .pio/build/esp32_4MB_max/src/wled.cpp.o
Compiling .pio/build/esp32_4MB_max/src/wled00.ino.cpp.o
In file included from wled00/../usermods/customeffects/usermod_v2_customeffects.h:14:0,
                 from wled00/usermods_list.cpp:189:
wled00/../usermods/customeffects/arti_wled.h: In member function 'float ARTI::arti_external_function(uint8_t, float, float, float, float, float)':
wled00/../usermods/customeffects/arti_wled.h:193:19: error: '__gnu_cxx::__alloc_traits<std::allocator<Segment> >::value_type {aka class Segment}' has no member named 'drawArc'
           SEGMENT.drawArc(par1, par2, par3, par4);
                   ^
wled00/../usermods/customeffects/arti_wled.h:195:19: error: '__gnu_cxx::__alloc_traits<std::allocator<Segment> >::value_type {aka class Segment}' has no member named 'drawArc'
           SEGMENT.drawArc(par1, par2, par3, par4, par5); //fillColor
                   ^
Compiling .pio/build/esp32_4MB_max/src/wled_eeprom.cpp.o
Retrieved `.pio/build/esp32_4MB_max/src/wled_math.cpp.o' from cache
*** [.pio/build/esp32_4MB_max/src/usermods_list.cpp.o] Error 1
Compiling .pio/build/esp32_4MB_max/src/wled_serial.cpp.o
========================================================================== [FAILED] Took 22.41 seconds ==========================================================================
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Fonta, 2023-03-10:
At this moment build fails even if custom effect is disabled.

[image]

[image]

MoonModules, 2023-03-28:
Hi Fonta,

This has been solved.
Can you check and can we close this issue?

Fonta, 2023-03-28:
@MoonModules Indeed seems to be solved. Cheers.

In the meanwhile the environments for codm are preventing a succesful built. It doesn't like the .
Error: Invalid environment name 'codm-controller-0.6-rev2'. The name can contain alphanumeric, underscore, and hyphen characters (a-z, 0-9, -, _)

closed 2023-03-28 by Fonta