MoonModules/WLED-MM #19: Esp32 c3

TysTies, 2023-02-16:
Hey, I just got a seed studio esp32 c3 off amazon and put weld on it. I was wondering if you could make a version for it. here is the Aircoookie Esp c3 stuff https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/S2_S3_C3

Thanks!

srg74, 2023-02-16:
Is Aircoookie firmware works for you?

softhack007, 2023-02-17:
Out c3 buildenv [env:esp32c3dev_4MB_max] could work for you, if you have
- at least 4MB Flash
- no PSRAM

softhack007, 2023-02-17:
> i just want a sound reactive version for esp 32 c3

I you already got WLED working on your -C3, then you should indeed use the MoonModules fork for audioreactive. I've recently made some optimization for -C3, so you should have more "fluent" animations, and less lagging behind and other problems with our version.

TysTies, 2023-02-18:
Could you send me a link? i can't find it.

softhack007, 2023-02-18:
> Could you send me a link? I can't find it. (To a bin file I can download)

Sorry, no pre-compiled bin availeable, so you'll need to build yourself in VSCode (platformio). Considering the experimental nature of -C3 support, this might any way be the better option. We have seem many problems with pre-compiled -C3 firmware that simply won't start (bootlooping).

Maybe @srg74 you could add a build of esp32c3d3v_4MB_max to your repo? @TysTies pre-built firmware may or may not work on -C3, it's not as simple install-and-go as on "classic esp32". If it does not run, then you need to build&upload yourself from VSCode.

srg74, 2023-02-19:
Done.

TysTies, 2023-02-23:
@srg74 Where would I find it?

srg74, 2023-02-23:
> @srg74 Where would I find it?

https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/%40MoonModules/v0.14.0-b2.19/WLEDMM_0.14.0-b2.19_esp32c3dev_4MB_max.bin

softhack007, 2023-02-24:
@TysTies please close this ticket if the bin above works for you.

softhack007, 2023-03-04:
Look like the issue is solved.

closed 2023-03-04 by softhack007