MoonModules/WLED-MM #21: Global I2C pin setting being overwritten

scottrbailey, 2023-02-17:
### What happened?

Saving changes on the 4LineDisplay config page overwrites the global I2C pin settings when "use global" option is used. Even though the correct pins are shown in "use global" drop down, before the save.

Rebooting after setting global I2C pins and before saving 4LineDisplay page does not make a difference. This bug doesn't impact upstream because all settings are on the same page.

And while your in the code here, maybe you could swap the order of the inputs so they match the order on Usermods page.  Usermods (Data, Clock), 4LineDisplay (Clock, Data).

### To Reproduce Bug

- set global I2C pins on usermods page
- save 4LineDisplay config page

### Expected Behavior

global I2C settings should be preserved.

### Install Method

From srg74 firmware repository

### What version/release of MM WLED?

WLEDMM_0.14.0-b1.18_esp32_4M_max

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ewoudwijma, 2023-02-17:
WLEDMM_0.14.0-b1.18_esp32_4MB_max is full of usermods, where each of the usermods can hijack global pins. It is work in progress to solve this. But this issue helps to prioritize this.

for the time being, try how WLEDMM_0.14.0-b1.18_esp32_4MB_max behaves, it has less usermods, and I think it works better here.

troyhacks, 2023-02-17:
I agree with both of these issues and have experienced the same - including the data/clock order confusion. 

I'll take a look at the code and see if there's any quick wins. The clock/data (and the labeling of them) should be a minor change. 

Other than that, I think we're into "pin registry" improvements.

scottrbailey, 2023-02-17:
> for the time being, try how WLEDMM_0.14.0-b1.18_esp32_4MB_max behaves

Do you mean WLEDMM_0.14.0-b1.18_esp32_4MB_min or a different version?

ewoudwijma, 2023-02-17:
No _max, see also https://mm.kno.wled.ge/moonmodules/Installing-and-Compiling/

_min, _max and _all is little confusion might update that to something more clear in the future

scottrbailey, 2023-02-17:
Oh, well that was the version that I was running.  So I'm playing one of those "spot the difference" games trying to figure out what you mean :)

I think any version with the 4LineDisplay mod enabled is going to exhibit that behavior. The settings page for that mod should not modify the value of hw>if>i2c-pin, only um>4LineDisplay>pin.

softhack007, 2023-02-17:
> The settings page for that mod should not modify the value of hw>if>i2c-pin, only um>4LineDisplay>pin.

Yes that look like a bug. I'll take a look at it next week, when I'm back from holidays. Also the "swapped order" should be simple to solve - you're right the normal order is SDA, SCL.

softhack007, 2023-03-20:
Current status:
-pin assignment: fixed
-order of SDA/SCL fields: to be fixed