MoonModules/WLED-MM #22: Webserver? crashing after some time

ReviloEgros, 2023-02-24:
### What happened?

The webserver or something related to it is crashing after some time. It happened to me changing effects, saving presets, turning on and off etc. The devices web interface is not reachable but the esp stays connected to wifi and is controllable through mqtt. If i force a wifi reconnection on my router for that device the webserver is reachable again.

### To Reproduce Bug

Just play around with settings, effects, presets etc. After some time the web interface isn't reachable anymore.

### Expected Behavior

Being able to open the web interface 100% of the time without my router forcing a wifi reconnection or to reboot my esp.

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

WLEDMM_0.14.0-b2.19 esp32_4MB_M.bin build 2302231

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-02-24:
Thank you for this report. 

1. Is this crash only happening in the MM fork, or does it also crash when you use ["upstream" WLED](https://github.com/Aircoookie/WLED)?
2. To better understand what is happening, please install a "debug" version of the MM firmware, Attach a USB cable, then watch serial output on your Computer. You can use PlatformIO serial Monitor, or Arduino IDE Serial Monitor for this.
Please attach the serial monitor output here, so we can further analyse your problem.
     *  [WLEDMM_0.14.0-b2.19_esp32_4MB_M_debug.bin.zip](https://github.com/MoonModules/WLED/files/10825465/WLEDMM_0.14.0-b2.19_esp32_4MB_M_debug.bin.zip)

softhack007, 2023-03-20:
Without additional information, it's hard to decide if your problem is related to some MoonModules specific features or not. As the webServer is not MM specific, your problem should also exists in upstream "vanilla" WLED.

Please open a new report for upstream.
https://github.com/Aircoookie/WLED/issues/new/choose

Closed for MM as not we cannot reproduce.

closed 2023-03-20 by softhack007