MoonModules/WLED-MM #23: Ethernet IP not being displayed, doesn't kick in without WiFi being cleared out (but working fine)

troyhacks, 2023-03-10:
### What happened?

Testing the Olimex ESP32-Gateway with ethernet and the latest code.

Ethernet (QuinLED-ESP32 matches this board's pins) doesn't kick in unless WiFi details are cleared out

After it starts working, it doesn't show any indication of the ethernet IP. (it used to) 

### To Reproduce Bug

Compile from source, select QuinLED-ESP32, and clear wifi details and reboot. 

### Expected Behavior

WiFi section should show "LAN IP" or similar if it's connected. 

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

WLEDMM 0.14.0-b2.19 (build 2303100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
---WLED 0.14.0-b2.19 2303100 INIT---
 WLEDMM_0.14.0-b2.19 esp32_4MB_M, build 2303100.
CPU:   ESP32-D0WDQ5 rev.3, 2 core(s), 240MHz.
CPU    Software restart via esp_restart() (code 3). Core#0 SW restart (12); Core#1 SW restart (12).
FLASH: 4MB, Mode 2, speed 80MHz.

Mount FS
done Mounting FS
deserializeConfig
initC: *** Ethernet successfully configured! ***
pinmgr not success for global i2c -1 -1
pinmgr not success for global spi -1 -1 -1
serializeConfig
Usermods setup ...
temperature usermod initialized.
AR: sound input driver initialized successfully.
mpu setup
mpu6050: warning - ivalid I2C pins: sda=-1 scl=-1
Ada

GPIO    | Assigned to           | Info
--------|-----------------------|------------
i/o   0   Button
i/o   1   ./.                     Serial TX
i/o   2   ./.
i/o   3   ./.                     Serial RX
i/o   4   ./.
i/o   5   Ethernet                (default) Rotary CLK pin
i/o  12   ./.
i/o  13   ./.
i/o  14   AudioReactive (UM)
i/o  15   AudioReactive (UM)      (default) mpu6050 INT pin
i/o  16   LEDs (digital)          (default) LED pin
i/o  17   Ethernet
i/o  18   Ethernet                (default) Rotary DT  pin
i/o  19   Ethernet                (default) Rotary SW  pin
i/o  20   ./.
i/o  21   Ethernet                (default) DHT temperature pin
i/o  22   Ethernet                
i/o  23   Ethernet
i/o  25   Ethernet
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
Client request
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

troyhacks, 2023-03-10:
Previously, if ethernet was connected it would override the wifi and show the ethernet IP.

Now it's not overriding wifi and not showing the ethernet IP even if it's working.

troyhacks, 2023-03-10:
[image]

softhack007, 2023-03-11:
oops, did not mean to close this.

ewoudwijma, 2023-03-29:
@troyhacks , is this MM specific or also in upstream?

troyhacks, 2023-05-02:
Still waiting on personal free time to re-test and cross-test this. :)

closed 2024-06-26 by troyhacks