MoonModules/WLED-MM #29: The 'MoonModules/WLED-Effects' repository doesn't contain the 'CustomEffects/wled' path in 'master'.

stiw47, 2023-03-30:
### What happened?

Hi guys. First of all, big thanks for the great mod! I'm not sure should this be reported here or on [WLED-Effects](https://github.com/MoonModules/WLED-Effects) repo, but anyway....
On [WLED-Effects](https://github.com/MoonModules/WLED-Effects) repo, name of the folder was changed from CustomEffects to ARTIFX, so download of custom effects code from Custom Effects Editor is not working any more.

### To Reproduce Bug

In the MM WLED web UI, go to Segment, then to Custom Effects Editor and click on Download button - nothing will happen. I figured out what is going on when I clicked on Custom Effects Library button:

[image]

==>

[image]

==>

[image]

### Expected Behavior

As you guess, custom effect code should be downloaded.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

 WLEDMM_0.14.0-b1.13_esp32_4MB_max.bin

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell
not relevant
```

### Anything else?

Thank you to the Moon and back for such a great fork!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

stiw47, 2023-03-30:
I'm not sure, I was downloading from [Releases](https://github.com/MoonModules/WLED/releases) page since now, last release is from December 2022 😁
If you could point me where from I could download it, I would be graceful. I believe you mean on daily builds? I read about that once on MM WLED wiki, but somehow I missed it out or didn't find it don't remember, most probably cause I was satisfied with release.

Thx.

lost-hope, 2023-03-30:
https://github.com/MoonModules/WLED/actions
click on the top most action and then scroll down to the artifacts. 
there are the default compiled enviroments

stiw47, 2023-03-30:
Didn't knew for this, thanks a lot for the info, closing.

closed 2023-03-30 by stiw47