MoonModules/WLED-MM #31: ArtiFX on ESP8266

lost-hope, 2023-04-06:
When adding the ArtiFX usermod to the env of the ESP8266, the main issue seems to be that the function `esp_get_free_heap_size()` is not present for ESP8266 (https://github.com/espressif/ESP8266_RTOS_SDK/blob/35375c17be0b7837239c330d9a411a0ad46196b8/components/esp8266/source/system_api.c#L378)
That function itself needs the `heap_caps_get_free_size` function (https://github.com/espressif/ESP8266_RTOS_SDK/blob/35375c17be0b7837239c330d9a411a0ad46196b8/components/heap/src/esp_heap_caps.c#L69)
That function uses two global variables, that need to be properly initialized: `g_heap_region_num` and `g_heap_region` most of it seems to be initialized here (https://github.com/espressif/ESP8266_RTOS_SDK/blob/35375c17be0b7837239c330d9a411a0ad46196b8/components/heap/port/esp8266/esp_heap_init.c#L21)

softhack007, 2023-04-06:
`ESP.getFreeHeap()` should also exists on 8266. It delivers the same information as esp_get_free_heap_size(), but it's part of the standard framework.

MoonModules, 2023-04-07:
ESP.getFreeHeap worked! And some other changes and I am able to compile it, but then the result of ESP.getFreeHeap is

setup 8552 bytes free
open /wledv033.json 8128 ✓
definitionTree 0 => 8088 ✓
deserializeJson() of definition failed with code NoMemory
Setup not successful

Around 8K free while on ESP32 around 100K free, and that is what ARTI-FX needs (maybe like 50k works...)

Is it possible to allocate more heap?

lost-hope, 2023-04-07:
yep seems to work:
I added a small ifdef at beginning of arti_wled.h:
```#ifdef ESP32
#define FREE_SIZE esp_get_free_heap_size() // ESP32 is set in wled context: small trick to set WLED context
#define ARTI_PLATFORM ARTI_ARDUINO         // else on Windows/Linux/Mac...
#endif

#ifdef ESP8266
#define ARTI_PLATFORM ARTI_ARDUINO // else on Windows/Linux/Mac...
#define FREE_SIZE ESP.getFreeHeap()
#endif```

and encased the `#define FREE_SIZE esp_get_free_heap_size()``in arti.h in a ifndef, so it does not override the other setting. 

i am getting an error for no `convertFromJson` function in ArduinoJson-v6. how did you manage to circumvent that?

MoonModules, 2023-04-07:
I did something similar (used ARDUINO_ARCH_ESP32/ESP8266 instead of ESP32/ESP8266), plus had to change the way strings are send to Serial/File. Will commit in half an hour I guess...

MoonModules, 2023-04-07:
Committed! https://discord.com/channels/700041398778331156/1060959116526362634/1093848593489739879

Please test both on esp8266 (likely crash due to low mem but maybe you can increase mem alloc some way) and on esp32 (using USER_PRINTF now, so ARTI-FX can also be tracked in Net Print)

MoonModules, 2023-04-07:
This asChar stuff is also something to look at, this caused me grey hairs last year when I made Arti, Serial.print looks so vulnerable for crashes, I got it now stable for ESP32 using .as<std::string>().c_str(), which uses the string type as 'bridge' but if I use the direct way: .as<const char *>() then it crashes on ESP32. For ESP8266 I now have turn it off completely (#define asChar(x) "", .as<std::string>().c_str() caused compile errors) so all ERROR_ARTI and DEBUG_ARTI will print nothing for asChar parts (to be on the save side) but would be nice if print

[image]

inddigital-xtech, 2024-10-06:
Hello, I don't know if it's a little late to comment here, but does this mean that ARTI-FX doesn't work on ESP8266?

Sorry I'm somewhat new to WLED but I've searched a lot and there are almost no references to this on the internet, https://www.reddit.com/r/WLED/comments/1fulodw/artifx_not_showing/

ewowi, 2024-10-06:
Never too late 🙂
But reading from above it might compile, but too slow and too memory demanding for esp8266
So use esp32.
Plus artifx will be replaced by asmParser in the future.
It is already working in StarLight: see the StarMod channel for more info.
This works so much better !

inddigital-xtech, 2024-10-09:
Ok, thank you very much for your response