MoonModules/WLED-MM #33: Maximum current and 2D configuration are reset on power cycle

stiw47, 2023-04-09:
### What happened?

I set up max current to 4500mA for my use case.
My LED matrix is 16x16, so this is also the setup in my 2D configuration.

After each power cycle, I can find that max current is changed to 5000mA and setup in my 2D configuration is changed to 8x8.
I didn't noticed so far that anything else is messing except above 2 things.

Firstly I mean, maybe is problem with me, cause I mess a lot with different presets, so maybe it pulling these values who knows where from, but I also tried with factory reset => fresh installation and basic manual config for 16x16. However, I'm still not sure is it ESP32 keep something even after factory reset, so I would be graceful if someone could try to reproduce this. Below video explains more than my post. Issue is recorded in (atm) last commit #500, but I noticed it before in #476 (maybe even before that), but I was lazy to sit here and type 😁

https://cloud.stiw47.cloud/s/wTATwEGP52Ny28y

### To Reproduce Bug

Power cycle LED matrix => check max current and 2D layout => compare it with previous values.

### Expected Behavior

It should keep settings between power cycles 😁

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

Commit #500 Change library for ArtiFX

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Unfortunately, I haven't it. Never catch a log for WLED before, I heard it is not that simple. On the other hand, I saw on MM repo there are some binaries with _debug in the file name, so if log is necessary, I can try.
```

### Anything else?

Thanks for great software.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

stiw47, 2023-04-09:
If someone else is experiencing such an issue, I'm work around'ing it with Home Assistant automation as for now:

```
- id: '1681036996891'
  alias: King Matrix Restore Conf
  description: ''
  trigger:
  - platform: state
    entity_id:
    - switch.light_switch_kids_2
    to: 'on'
  condition: []
  action:
  - wait_for_trigger:
    - platform: state
      entity_id:
      - light.king_matrix
      to: 'on'
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - service: rest_command.king_matrix_restore_conf
    data: {}
  - delay:
      hours: 0
      minutes: 0
      seconds: 2
      milliseconds: 0
  - service: button.press
    data: {}
    target:
      entity_id: button.king_matrix_restart
  mode: single
- id: '1681037968114'
  alias: Light Switch Kids 2 Off King Matrix Off
  description: ''
  trigger:
  - platform: state
    entity_id:
    - switch.light_switch_kids_2
    from: 'on'
    to: 'off'
  condition: []
  action:
  - service: light.turn_off
    data: {}
    target:
      entity_id: light.king_matrix
  mode: single
```

Rest command:

```
king_matrix_restore_conf:
  url: "http://192.168.0.188/json/cfg"
  method: POST
  content_type: "application/json"
  payload: '{"rev":[1,0],"vid":2304080,"id":{"mdns":"king-matrix","name":"King Matrix","inv":"Light"},"nw":{"ins":[{"ssid":"Please wait void...","pskl":19,"ip":[0,0,0,0],"gw":[0,0,0,0],"sn":[255,255,255,0]}]},"ap":{"ssid":"WLED-AP King Matrix","pskl":8,"chan":1,"hide":0,"behav":0,"ip":[4,3,2,1]},"wifi":{"sleep":false},"hw":{"led":{"total":256,"maxpwr":4500,"ledma":55,"cct":false,"cr":false,"cb":0,"fps":42,"rgbwm":255,"ld":false,"matrix":{"mpc":1,"ba":true,"mph":1,"mpv":1,"pb":false,"pr":false,"pv":false,"ps":false,"panels":[{"b":false,"r":false,"v":false,"s":true,"x":0,"y":0,"h":16,"w":16}]},"ins":[{"start":0,"len":256,"pin":[4],"order":0,"rev":false,"skip":0,"type":22,"ref":false,"rgbwm":0}]},"com":[],"btn":{"max":4,"pull":true,"ins":[{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]}],"tt":32,"mqtt":false},"ir":{"pin":-1,"type":0,"sel":true},"relay":{"pin":-1,"rev":false},"baud":1152,"if":{"i2c-pin":[-1,-1],"spi-pin":[-1,-1,-1]}},"light":{"scale-bri":100,"pal-mode":0,"aseg":false,"gc":{"bri":1,"col":2.8,"val":2.8},"tr":{"mode":true,"dur":7,"pal":0,"rpc":5},"nl":{"mode":1,"dur":60,"tbri":0,"macro":0}},"def":{"ps":16,"on":true,"bri":255},"if":{"sync":{"port0":21324,"port1":65506,"recv":{"bri":true,"col":true,"fx":true,"grp":1,"seg":false,"sb":false},"send":{"dir":false,"btn":false,"va":false,"hue":true,"macro":false,"grp":1,"ret":0}},"nodes":{"list":true,"bcast":true},"live":{"en":true,"mso":false,"port":5568,"mc":false,"dmx":{"uni":1,"seqskip":false,"e131prio":0,"addr":1,"dss":0,"mode":4},"timeout":25,"maxbri":false,"no-gc":true,"offset":0},"va":{"alexa":false,"macros":[0,0],"p":0},"mqtt":{"en":false,"broker":"","port":1883,"user":"","pskl":0,"cid":"WLED-5155f4","topics":{"device":"wled/5155f4","group":"wled/all"}},"hue":{"en":false,"id":1,"iv":25,"recv":{"on":true,"bri":true,"col":true},"ip":[192,168,0,0]},"ndb":{"ip":[0,0,0,0],"port":1768,"enabled":false},"ntp":{"en":false,"host":"0.wled.pool.ntp.org","tz":2,"offset":0,"ampm":false,"ln":0,"lt":0}},"ol":{"clock":0,"cntdwn":false,"min":0,"max":29,"o12pix":0,"o5m":false,"osec":false},"timers":{"cntdwn":{"goal":[20,1,1,0,0,0],"macro":0},"ins":[{"en":1,"hour":255,"min":0,"macro":0,"dow":127},{"en":1,"hour":255,"min":0,"macro":0,"dow":127}]},"ota":{"lock":false,"lock-wifi":false,"pskl":7,"aota":true},"um":{"Temperature":{"enabled":false,"pin":18,"degC":true,"read-interval-s":60,"parasite-pwr":false,"parasite-pwr-pin":-1},"4LineDisplay":{"enabled":false,"pin":[-1,-1,-1,-1,-1],"type":1,"flip":false,"contrast":10,"contrastFix":false,"screenTimeOutSec":60,"sleepMode":true,"clockMode":false,"showSeconds":true,"i2c-freq-kHz":400},"Rotary-Encoder":{"enabled":false,"DT-pin":-1,"CLK-pin":-1,"SW-pin":-1,"preset-low":0,"preset-high":0,"apply-2-all-seg":true},"Autosave":{"enabled":false,"autoSaveAfterSec":15,"autoSavePreset":250,"autoSaveApplyOnBoot":false},"AudioReactive":{"enabled":false,"analogmic":{"pin":-1},"digitalmic":{"type":1,"pin":[32,15,14,-1,-1,-1]},"config":{"squelch":10,"gain":60,"AGC":1},"experiments":{"micLev":0,"freqDist":0,"freqRMS":1},"dynamics":{"limiter":true,"rise":50,"fall":300},"frequency":{"scale":3,"profile":0},"sync":{"port":11988,"mode":0}},"Weather":{"enabled":false,"apiKey":"","units":1,"minTemp":0,"maxTemp":40},"mpu6050-IMU":{"enabled":false}}}'

```

dosipod, 2023-04-10:
You have not listed the MM version used . please use latest build  from here https://wled-install.github.io/ , which i have tested with your setting and did not notice any issue . Once you are on latest MM you can also use net print ( net debug ) .

[image]
[image]

stiw47, 2023-04-10:
Oh, sorry cause MM WLED version. One more thing I didn't know that exist at all - https://wled-install.github.io/ - thanks for this info. Previously, I installed MM WLED by going to MM WLED github => Actions => latest action => pick appropriate version from Artifacts (ESP32 4MB M). So as mentioned before in my post, it was action called Change library for ArtiFX, number 500.

Just tried to install WLEDMM_0.14.0-b15.22_esp32_4MB_M.bin from link provided by you and I have same behavior - layout reset to 8x8 and max amps reset to 5000mA. Will try tonight to figure out where to enable and how to use net debug, and will follow up.

dosipod, 2023-04-10:
If you used the bin from site then for sure that is working with us .I would recommend that you fresh install with USB and not OTA if you are not doing so already . For netdebug , look at sync screen pic above and just add the ip of your PC and then run nc -l 1768 -u from that PC .  As said this is working with us and i have kept the unit since last night with netdebug active and did not notice anything .
Small things i am not sure about if you are still facing the issue without physical leds attached . And why in reality you need that much power for 16x16 matrix ? if that is a ready made matrix of ws2812b then having 2.5 or 3 A is enough . If you will post video then try to make it shorter   or use  pics of the setup but long video is hard make and hard to follow :)

stiw47, 2023-04-10:
Hi, thanks for answer.

> If you used the bin from site then for sure that is working with us

Yes, I used bin from the site this time. If you are confirming it is working ok on your and, then probably something on my end surviving OTA, I'm closing this now.

> I would recommend that you fresh install with USB and not OTA if you are not doing so already 

I'm installing it via OTA. The matrix is packed in wooden box and mounted on ceiling with screws. It has also basic non addressable white strip on border, so it can be ordinary ceiling light in kids room, together with WLED effects on matrix (one ESP32) and Tetris/Snake/Breakout games on same matrix controlled by PS4 controller (another ESP32) (yeah, I know, washing machine is probably missing here 😂). Since everything is hand made by me (even wooden box), it is no so simple to unmount/mount it, and USB flashing is no option for me until potential serious break or damage. Closing issue now and will live with HA automation I already made.

> For netdebug , look at sync screen pic above and just add the ip of your PC and then run nc -l 1768 -u from that PC

Thanks a lot for this info, didn't know for this as well. Unfortunately, not working for me, I have no any output info in console, neither from Windows 10 nor from Archlinux. Tried to execute a command as both - normal user and root/admin. Will play with this more tomorrow.

[image]

[image]

> Small things i am not sure about if you are still facing the issue without physical leds attached

Hmmmm, in order to answer this question, I tried to change output data PIN from 4, which I'm btw using, to 16, where nothing is attached. However, after power cycling device, I found that my data output PIN is magically set to 4 again. I now believe more this is some issue on my end and definitely closing this, sorry cause false alarm.

> And why in reality you need that much power for 16x16 matrix ? if that is a ready made matrix of ws2812b then having 2.5 or 3 A is enough 

This is not relevant I believe. I set it to 4.5A cause my PSU is 5A, but thanks for advice anyway.

closed 2023-04-10 by stiw47

softhack007, 2023-04-10:
My best guess on this problem is corrupted flash filesystem, or possibly a full filesystem that was created too small. As a consequence, you could change settings, but WLED would fail to write them into files, so at the next startup your changes seem to be lost.

Maybe you initially uploaded WLED with ESP home Flasher. This tool is known for creating borked filesystem - see WLED knowledge base.

Unfortunately an OTA update cannot repair corrupted flash filesystems. You'd need to wipe your device via USB, then re-install via USB to fix this kind  of problem.

stiw47, 2023-04-11:
If anybody is interested yet, I found it. Issue was on my end (of course 😶‍🌫️) and in best words, I am an idiot 😁. Sorry cause confusion and thank you for your time.

I mentioned before, this matrix is combo of WLED (one ESP32) and FastLED Tetris/Snake/Breakout (another ESP32). Both ESP's32 are connected to same data IN PIN on LED matrix. In order to be able to handle this without interference, and to use only one at given moment depending on use case, I had to write some code. For WLED board enable/disable, I didn't found any other way except that I change it's output data PIN. E.g. if WLED board should be disabled, there is a class in Arduino FastLED code, which sending configuration to WLED board via REST command, JSON API, to change it's output data PIN to some dummy PIN, where nothing is connected. And vice versa for use case when WLED board should be enabled => set PIN to GPIO4, where matrix is connected. In this REST command, I'm sending whole conf and this was a problem (sorry guys, I made this few months ago and forgot it). This is the class:

```
// APP_WLEDREAL

class WledReal {
  
  public:

  void setup() {
      HTTPClient http;
      http.begin(espClient, wledCfg);
      http.addHeader("Content-Type", "application/json");
      int httpResponseCode = http.POST("{\"hw\":{\"led\":{\"total\":256,\"maxpwr\":5000,\"ledma\":55,\"cct\":false,\"cr\":false,\"cb\":0,\"fps\":42,\"rgbwm\":255,\"ld\":false,\"matrix\":{\"ph\":16,\"pw\":16,\"mph\":1,\"mpv\":1,\"pb\":false,\"pr\":false,\"pv\":false,\"ps\":false,\"panels\":[{\"b\":false,\"r\":false,\"v\":false,\"s\":true}]},\"ins\":[{\"start\":0,\"len\":256,\"pin\":[4],\"order\":0,\"rev\":false,\"skip\":0,\"type\":22,\"ref\":false,\"rgbwm\":0}]}}}");
      Serial.print("HTTP Response code: ");
      Serial.println(httpResponseCode);
      http.end();
  }

  boolean loop() {
  }
  
};
```

So now is clear why my Maxpower was mess back to 5000mA, and why data PIN was mess back to GPIO4. Regarding matrix layout, mess back from 16x16 to 8x8, idk. I can see above in my REST command it is already set to 16x16. On the other hand I can see in WLED UI there is noticeable difference in 2D Configuration panel between WLED official and WLED MM, and this above was initially coded for WLED official, few months ago when I didn't know that WLED MM exist at all. So I can only guess that JSON string should be made differently for WLED MM and that in my case it falls back to 8x8, which is if I remember correct default layout (didn't catch the time to test this yet, working hours 😒).

@dosipod @softhack007 : Thank you very much for prompt responses and sorry once again for confusion.