MoonModules/WLED-MM #38: Cannot build for ESP32C3

Christanoid, 2023-04-29:
### What happened?

I used PlatformIO (also tried fully reinstalling PlatformIO including clearing all files at my user directiory/.platformIO, however I cannot get this to compile for either the seeed_esp32_c3_4MB_S or the esp32c3dev_4MB_M environments. Instead I always get the following message:

`Platform Manager: espressif32@5.1.1 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'
`

### To Reproduce Bug

Download current master branch or MoonModules v0.14.0-b15, select seeed_esp32_c3_4MB_S or the esp32c3dev_4MB_M environments, try to build.

### Expected Behavior

A successful compile ;)

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

seeed_esp32_c3_4MB_S or esp32c3dev_4MB_M

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-04-29:
Hi, yes its  known problem. Neither MM nor upstream WLED can be build at the moment for -C3, -S2, or -S3. It seems that espressif have messed up something in their toolchain repo.

Also we cannot "simply" use their newest platforms, due to instability problems observed on -C3 and -S2.

So it seems you have to wait for a few weeks, sorry 🤷

Christanoid, 2023-04-29:
Hm, I see. Well guess i'll have to wait with my project, the precompiled C3.bin files I have been able to find unfortunately all bootloop on my Seeed Xiao Esp32c3.

softhack007, 2023-04-30:
There is a suggestion on how to fix it here, I'll give it a try in the next days.

* https://github.com/platformio/platform-espressif32/issues/1081#issuecomment-1518601054

softhack007, 2023-04-30:
It works now on github, can you try again on your PC?

Christanoid, 2023-05-01:
Am currently on a family trip, will verify once back home later today!

Christanoid, 2023-05-01:
Can confirm that it works with the patch you linked earlier. Now I only need to figure out the error in my Wiring that makes the LEDs flicker like crazy when using a logic level shifter and slightly when directly controlling from controller 😆

closed 2023-05-01 by Christanoid

troyhacks, 2023-05-02:
> Can confirm that it works with the patch you linked earlier. Now I only need to figure out the error in my Wiring that makes the LEDs flicker like crazy when using a logic level shifter and slightly when directly controlling from controller 😆

Add a 50Ω to 100Ω resistor to your data line(s).  That often sorts random flickering out. 

Deep-dive explanation: https://www.youtube.com/watch?v=H4pFvqIxkhQ