MoonModules/WLED-MM #42: PDM Mic works great and so cheap price

DJprasenjit, 2023-05-08:
hello 
thanks for this fork. 
i using in a 16x16 matrix
and i used ADC pin to analog sound input. its works fine. but i saw some videos and search in wled wiki and found out that its works so good with I2S mics
but there is some modules like INMP441 in amazon very costy. here india its 400INR 
then i found out a new trick and its costs me only 10INR (40 times less price)
I work mobile repair and hardware. then i found a mic that used in mostly new sound system mobile phone and its also a I2S _PDM mic
Then i visit my local market and search for it, after searching a while i found a mic that used for LYF F320B model . i got 10 Pc Mic at only 100INR

then i searched some datasheet and found the connections and i Surprised its worked. and works so good, no noise no humming . i tested the 100hz to 5Khz frequency and its working good, so i suggest to go for it. 
If anyone interested to buid contact me i will help
telegram @DJPrasenjit

dosipod, 2023-05-08:
Sure we are interested , post a  link to the mic

softhack007, 2023-05-08:
I2S PDM is already supported and possible on ESP32 and ESP32-S3.
Are you interested in better support for a specific microphone model?

DJprasenjit, 2023-05-09:
https://youtube.com/shorts/SGwSfZxMjRc
Custom made WLED with this fork

dosipod, 2023-05-09:
Yes wled is great for matrix  , i am not sure why the title of "PDM Mic works great and so cheap price"  , if you are selling HW to open source community then please try to be transparent