MoonModules/WLED-MM #43: Weird bin selection behaviour?

Fonta, 2023-05-08:
### What happened?

Not sure if this behaviour is to be expected?
Also happens with E.G. Puddlepeak where you can select a bin.

See gif:
[image]

### To Reproduce Bug

Select a effect where you can set a bin, try to set bin to 0.

### Expected Behavior

To stay on 0?

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

WLEDMM_0.14.0-b15.23_esp32_4MB-max (build 23050700)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-06-07:
Technicially speaking, selecting "bin 0" or "volume 0" does not make any sense. These settings control the peak detect algorithm. "Bin 0" is the 0hz FFT channel (no beat), "volume 0" would mean that silence is counting as a beat.

The observed behaviour is actually expected, as the effect tries to restore "meaningful values" when the user wants something impossible.

Closing as this not a bug.

closed 2023-06-07 by softhack007

Fonta, 2023-06-18:
@softhack007 the effect only seems to work when bin is set to 0. But now that doesn't work anymore, the whole ripple peak and puddlepeak are broken as they are producing 0 output. It doesn't matter which setting I put into the sliders, there's simply nothing happening.

softhack007, 2023-06-18:
I took a second look at the source code.
* there was a small typo (custom1 vs custom2) that caused the wrong slider to jump back to 48 - RipplePeak effect.
* Actually there are two peak detectors in the MM fork, however both are very bad at detecting beats. A new beat detector in "in the making", but needs a few more months before its ready for use.

1. the "FFT based" detector (which is the worst) was disabled - I've just re-enabled it because something bad might still be better than nothing. This detector is used when `bin>4 and minVolume>1`.
2. the "volume based" detector often misses a beat, but is more reliable to detect peaks. This detector is used when  bin >0 and bin < 10.

Setting bin = 0 or minVolume=0 is still considered a stupid idea, so the code does not support this.
You should however see effect reactions (not necessarily in sync to the music) when you set bin >2.

Fonta, 2023-06-18:
Thanks for checking and fixing the code @softhack007.
Now with bin set to 40 and min volume to 48 the effect is pretty ok.
Still when going above +/- 55 with the bin setting the amount of output drastically drops off.
Looking forward to the new beat detection.

Do I understand the following correctly:
FFT based is used if:
bin higher than 4
minVol higher than 1
Vol based is used if:
bin is higher than 0 but lower then 10

So with bin 40 and minvol on 48, it would be using the FFT based?
Only changing the bin to e.g. 8 would change it to volume based.
I've tried this and the output becomes really bus even with the fade rate slider all the way to minimum, there's practically no ripple effect anymore.

softhack007, 2023-06-18:
Maybe I was not explicit in my previous responses - the current beat detector is crap. No matter if FFT based or not. This applies to all forks (MM, AC, WLED-SR).

We can detect some peaks, which may be related to beats, or not. Broken beyond repair. 

Bin=40 means you're checking for "beats" around 1600hz. The normal "bass range" is between 60hz and 240hz. If you like what the effect does with this settings, that's fine. Just it's definitely not reactiveng to beats.

> Now with bin set to 40 and min volume to 48 the effect is pretty ok.
> Still when going above +/- 55 with the bin setting the amount of output drastically drops off.

My recommendation is to stay below 10 (bin 10=420hz)