MoonModules/WLED-MM #45: Analogue microphone audio reactivity not working with Lolin32lite

gaijinsr, 2023-05-15:
### What happened?

Thanks for this great WLED branch, I really like all the extra functionality! I have an analogue microphone connected to VP/IO36 on my Lolin32lite. With the SR branch binary (soundReactive_WLED_0.13.3_ESP32.bin) audio reactivity works just as expected.
With MM, there is no audio reactivity. Although I have not modified the hardware setup at all, the system reacts as if no audio is received. I have set the analogue input to IO36, the digital microphone type to analogue and all the digital pins to n/a. I am using WLEDMM_0.14.0-b15.21_esp32_4MB_M.bin and everything else works as expected. Any ideas?

### To Reproduce Bug

Install WLEDMM_0.14.0-b15.21_esp32_4MB_M.bin on a Lolin32lite, connect an analogue microphone to VP/IO36. No sound reactivity (although the same setup works with SR 0.13.2).

### Expected Behavior

The system should react to sound.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b15.21_esp32_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Thank you for your great work!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gaijinsr, 2023-06-02:
This still persists... Any idea, anybody?

softhack007, 2024-07-13:
This seems to be a hardware problem.
Lolin32lite seems to be an esp32, and in general analog input via gpio36 does work on esp32.

closed 2024-07-13 by softhack007