MoonModules/WLED-MM #47: Default power-on is invalid.

bluesubway, 2023-05-25:
### What happened?

After flashing the firmware, all functions are normal, but in the configuration and LED settings, check the LED box after power-on reset. 

### To Reproduce Bug

After disconnecting and reconnecting the power supply.the default power is still off.

### Expected Behavior

NULL

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b15.21_esp32_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-06-07:
The feature to turn LEDs on (or not on) after power-on is from upstream, and not specific to the MoonModules fork.

Please check if it works for you in upstream WLED, and create a new issue report upstream in case the problem also exists there.

https://github.com/Aircoookie/WLED/issues/new/choose

softhack007, 2023-06-07:
Closing, as this is not a feature that is specific to the MoonModules fork.

closed 2023-06-07 by softhack007