MoonModules/WLED-MM #53: Scrolling text: support for arbitrary date/time formats

softhack007, 2023-06-07:
### Is your feature request related to a problem? Please describe.

The WLED "scrolling text" effect has a few fixed formats to display date or time. Unfortunately people have diverging preferences on how a "digital clock" should look like, plus there are different formats that are considered "normal way" in each nation.

### Describe the solution you'd like

The solution could be to allow userdefined time/date formats from the "strftime()" standard function.

https://cplusplus.com/reference/ctime/strftime/

For example, naming a segment "%I:%M%p" would result in "03:21PM" (leading zeros, capital letter AM/PM).

### Additional context

This is an idea that came up in a discussion on discord.
It would be rather simple to implement, because strftime() seems to be availeable on esp32, and WLED already has a feature to convert internal "unixtime" to the tm struct needed by the function.

Thank you for your ideas for making WLED better!

dosipod, 2023-06-09:
Interesting , that would mean you could have day or week in there ?
[image]
And might not be linked to the post about time but rather general limitation with text is that its only ASCII

softhack007, 2023-06-29:
Some docs on the "system time" api that also supports `strftime()` and `strict tm`

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html#get-current-time

```c++
time_t now; 
char strftime_buf[64]; 
struct tm timeinfo;
  time(&now); // Set timezone to China Standard Time 
  setenv("TZ", "CST-8", 1); 
  tzset();
  localtime_r(&now, &timeinfo); 

strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo); 

ESP_LOGI(TAG, "The current date/time in Shanghai is: %s", strftime_buf);
```