MoonModules/WLED-MM #54: ESP32_4MB_XL doesn't have Animartrix

FragMenthor, 2023-06-08:
### What happened?

The XL version of the bins still don't have Animartrix MOD enabled

### To Reproduce Bug

Install XL version instead of M.

### Expected Behavior

The MOD should show on both versions

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

WLEDMM_0.14.0-b15.27 esp32_4MB_XL.bin build (2306070)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-07-28:
Hi,
unfortunately we had to remove animartix from `esp32_4MB_XL` as it does not fit into flash memory together with all other usermods. Sorry, fact of life - you cannot put more into memory than the memory chip can store..

As workaround, these builds include animartix:
- `esp32_4MB_M`
- `esp32_16MB_XL`

FragMenthor, 2023-07-30:
Ok, thanks for the clear answer! No problem.

closed 2023-07-30 by FragMenthor