MoonModules/WLED-MM #61: ES8388 - AiThinker - broken

netmindz, 2023-07-28:
### What happened?

I'm getting silence. Another user is only capturing from mic.

I'm getting no audio passthru 

### To Reproduce Bug

Linein on AiThinker v2.2

### Expected Behavior

Line-in

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

B25

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2023-08-11:
https://github.com/MoonModules/WLED/pull/48

softhack007, 2023-10-07:
@netmindz I think we can close this now?

netmindz, 2023-10-08:
Let me just double check with this board as well as the LyraT

netmindz, 2023-10-08:
Just checked and no error, but silence and no passthru, need to double check the config is right

netmindz, 2023-10-08:
PIN Manager: wire.begin successfull! I2C bus is active. SDA=33 SCL=32.
Starting display (I2C HW).
AR: ES8388 Source (Line-In)
ES8388Source:: initialize();
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x8, val 0x0).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x2, val 0xF3).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x2B, val 0x80).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x0, val 0x5).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0xD, val 0x2).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x1, val 0x40).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x3, val 0x0).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x4, val 0xFC).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x2, val 0x40).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0xA, val 0x50).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x9, val 0x0).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x10, val 0x40).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x11, val 0x40).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x26, val 0x9).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x27, val 0x50).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x2A, val 0x50).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x2E, val 0x1B).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x2F, val 0x1E).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x30, val 0x1E).
AR: ES8388 I2C write failed with error=2  (addr=0x10, reg 0x31, val 0x1E).
I2SSource:: initialize().

softhack007, 2023-10-08:
> Starting display (I2C HW).
This one comes from four line display. It should not cause any problem, but you could check if disabling the 4LD_alt makes a difference.

softhack007, 2023-10-08:
i2c error #2 is NACK - either the Es8388 address (0x10) is wrong, or the I2C pins ( SDA=33 SCL=32) are wrong/swapped.

softhack007, 2023-10-08:
> i2c error 2 is NACK when sending the address - either the Es8388 address (0x10) is wrong, or the I2C pins ( SDA=33 SCL=32) are wrong/swapped.

Update: I2C config seems to be correct, at least according to information I found here: https://www.pschatzmann.ch/home/2021/12/06/the-ai-thinker-audio-kit-experience-or-nothing-is-right/

So maybe its really some interference with four-line-display drivers. un-checking "enabled" in the fourline-display usermod setting, then reboot, might help.

netmindz, 2023-10-08:
If the initialisation fails, we need that to be showing on the info page, not just silence

softhack007, 2023-10-08:
> If the initialisation fails, we need that to be showing on the info page, not just silence

good point, I'll add a message tomorrow. I'll also check if a "frozen" message (discussed some month ago) can be added easily.

netmindz, 2023-10-08:
I just  tried the alternative pin combo for I2C on some boards and I get

AR: ES8388 Source (Line-In)
ES8388Source:: initialize();
PIN Manager: wire.begin successfull! I2C bus is active. SDA=18 SCL=23.
I2SSource:: initialize().
AR: I2S#0 driver without aPLL; fixed_mclk=0.
AR: 32 bits, Sample scaling factor = 1.0000
AR: I2S#0 driver installed in MASTER mode.
AR: FFT task started on core 0 [prio=1, min free stack=|| loopTask  min free stack 4360
AR: sound input driver initialized successfully.  

But still then silence and still no audio passthru

netmindz, 2023-10-08:
Might be useful

https://github.com/muvox-io/euphonium/blob/e26ee70af16602ae4bed48d94aca14974b988a19/euphonium/scripts/esp32/dacs/es8388_driver.be

softhack007, 2023-10-08:
> I just tried the alternative pin combo for I2C on some boards and I get
> But still then silence and still no audio passthru

Looks like a step forward - no i2c errors means that AR could talk to the chip 👍

netmindz, 2023-10-08:
I'm annoyed i never took a screenshot of the setting when  I had my board working, the notes on the docs are from @troyhacks and this board is notorious for having 3 variants

softhack007, 2023-10-08:
Maybe this table helps: https://github.com/pschatzmann/arduino-audiokit/wiki/GPIO-Overview-by-Selected-Board

[image]

netmindz, 2023-10-09:
Ah, great find @softhack007

netmindz, 2023-10-09:
[image]

closed 2023-10-09 by netmindz

dosipod, 2024-05-05:
Even if this is closed might be the best place to post about other variant  ( not sure what is marked on your board )  but this one is marked as ESP32 Audio kit V2.2  A404 
[image]