MoonModules/WLED-MM #68: Backing up config & presets .. no extension added

jakelight, 2023-08-31:
### What happened?

When you go to backup presets, or config .. Wled does not add the .json extension to file being saved.

you can either .. when saving, manually , in save as dialog box, change format to All Files, then manually add the .json, to make file(s) able to be loaded later ..

Or, you can rename the files, adding the ,json, from the desktop, to make them work.

So, this means the files are being saved in json format, it's just that Wled is not adding the extension when saving.

### To Reproduce Bug

Save your presets or config in ...

Config ---> Security & Updates

### Expected Behavior

That the files would have the proper ,json extension.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b27.31_esp32_4MB_XL.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jakelight, 2023-08-31:
Note: This was on a Up to date, Windows 11 machine

softhack007, 2023-12-30:
Closing, because I cannot reproduce the problem.
Possibly a mistake in windows settings, or IE configuration issue.

closed 2023-12-30 by softhack007