MoonModules/WLED-MM #70: Individual areas flicker

MOVE1904, 2023-09-04:
### What happened?

When I mix segments with different effects, individual areas flicker. The error appears in different configurations.
[image]

### To Reproduce Bug

Configuration 1: 
ESP32
Maximum current: 4000mA
GPIO  2:  42 LEDs
GPIO 19:  26 LEDs
GPIO 18: 250 LEDs
GPIO 21:  24 LEDs
GPIO  4: 178 LEDs

Segment 0:   0 -  42
Segment 1:  42 - 150
Segment 2: 150 - 229
Segment 3: 229 - 318
Segment 4: 318 - 342
Segment 5: 342 - 520

Segment 0: Effect: Solid           Color Palette: default Fx=white
Segment 1: Effect: Solid           Color Palette: default Fx=blue
Segment 2: Effect: Solid           Color Palette: default Fx=blue
Segment 3: Effect: Wavesins    Color Palette: default Fx=blue
Segment 4: Effect: Solid           Color Palette: default Fx=white
Segment 5: Effect: Wavesins    Color Palette: default Fx=white

In this configuration the LEDs 69 - 150 (parts of the solid segments) flicker even on the display.
With WLED AC there is no problem.

### Expected Behavior

I expected the solid segments solid.

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

build 2308250 WLED v0.14.0-b27.31 "This is the way"

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Maybe an error in brightness limiter?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-09-06:
It seems that the "flicker" is visible in the preview, so I can exclude its a problem with the NeoPixelBus driver.

My best guess right now is the "auto brightness limiter" feature is under stress to keep power consumption below the limits you set.

Please try the following options:
1) reduce global brightness (slider) 
2) increase "Maximum current: 4000mA"
and check if one of them reduces / removes flickering.

MOVE1904, 2023-09-06:
Here is another configuration with 2192 LEDs connected to 8 GPIOs, 5000mA maximum current and 7 segments.
As You can see the flicker stops by setting the brightness down.
You can also change the behavior by increasing the maximum current.

This behavior is only visible in the WLED MM Version. WLED AC works fine even with much lower maximum current (tested with 5000mA and 2000mA). There is no flicker, just darker with 2000mA.
[image]

softhack007, 2023-09-06:
> Here is another configuration with 2192 LEDs connected to 8 GPIOs, 5000mA maximum current and 7 segments.

@MOVE1904 could you  make a backup of your presets (presets.json) and config (cfg.json), and upload them here ?

It would help a lot to reproduce your problem.

MOVE1904, 2023-09-07:
O.K. I Have 2 for You.

[wled_presets.zip](https://github.com/MoonModules/WLED/files/12544668/wled_presets.zip)

Finnlife, 2025-01-23:
What's the eta on this?
The flicker really sucks

softhack007, 2025-01-23:
@Finnlife what's "eta" ? (sorry not a native speaker)

Finnlife, 2025-01-23:
@softhack007 sorry, here you go: https://en.wikipedia.org/wiki/Estimated_time_of_arrival
tl;dr
Just means "when this will be resolved?" (in this context)
Feel free to ask me if you have any question or need any information to analyze/debug this issue.

softhack007, 2025-01-23:
> The flicker really sucks

In case you have the same problem as described by @MOVE1904, it's most likely an issue with the brightness limiter being under stress to hold the current below limit. The solution can be:
* get a more powerful PSU (3A or so it really under-powered for >2000 leds), and make sure your cables are thick enough
* reduce overall brightness (slider)
* set a higher maximum current in LEDs settings (this will get the brightness limiter out of the "stress zone")

@Finnlife its true we use a slightly different code for the brightness limiter (different from upstream), that usually performs better. However right now I cannot give you any estimation for improvements of the "low power psu" behaviour, sorry 🤷. Generally it's better to invest a few coins for improving your electrical setup, rather than relying on a software feature to protect your roof from catching fire.