MoonModules/WLED-MM #74: ARTI-FX bricks controller

protyposis, 2023-09-28:
### What happened?

When setting an effect script in the ARTI-FX editor with "Save and Run", the controller sometimes crashes (after the first `.` is printed to the "Compile and Run Log") and ends up in a boot loop. To fix it, I need to reflash and reconfigure WLED.

### To Reproduce Bug

It seems to depend on the number of (global?) variables and/or the number of conditions, but I wasn't able to exactly pinpoint. Here's a simple repro:

```
program CrashRepro
{
  fr01 = 0
  fr02 = 0
  fr03 = 0
  fr04 = 0
  fr05 = 0
  fr06 = 0
  fr07 = 0
  fr08 = 0
  fr09 = 0
  fr10 = 0
  fr11 = 0
  fr12 = 0
  fr13 = 0
  fr14 = 0
  fr15 = 0
  fr16 = 0
  fr17 = 0
  fr18 = 0
  fr19 = 0
  fr20 = 0
  fr21 = 0
  fr22 = 0
  fr23 = 0
  fr24 = 0
  fr25 = 0
  fr26 = 0
  fr27 = 0
  fr28 = 0
  fr29 = 0
  fr30 = 0
  fr31 = 0
  fr32 = 0

  function renderFrame()
  {

  }
}

```

### Expected Behavior

No bricking, i.e., graceful error handling.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b27.31_esp32_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

[... 255 lines cut ...]

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a818:0x3ffb1dd0 0x4008ab75:0x3ffb1df0 0x4008fc79:0x3ffb1e10 0x40085fca:0x3ffb1e30 0x40086395:0x3ffb1e50 0x4000bec7:0x3ffb1e70 0x401d307d:0x3ffb1e90 0x4012caad:0x3ffb1eb0 0x4012efe1:0x3ffb1ee0 0x400ede77:0x3ffb1f30 0x40133032:0x3ffb1f50 0x401335c7:0x3ffb1f90 0x4015711d:0x3ffb1fb0 0x4008c5fe:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c

---WLED 0.14.0-b27.31 2309170 INIT---
[...loop continues...]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-09-29:
@ewoudwijma could you take a look?
It seems like an empty renderFrame program is causing some infinite loop of error messages in the destructor.
```
closed Arti 110344 
✓ Destruct ARTI 
Error: parseTree should be array or object null (9)
```

protyposis, 2023-09-29:
Seems like I simplified the repro case too much. The issue I described also happens with valid non-empty programs.