MoonModules/WLED-MM #79: Mixed mode 2D / 1D works upstream but not in MM

scottrbailey, 2023-10-09:
### What happened?

In AC you can do mixed 2D/1D configurations by adding a new segment beyond the bounds of your 2D matrix. With MM, when you add a new segment and set the start X beyond the last pixel of your matrix, the UI correctly switches to 1D segment.  However, once saved, it reverts back to 2D.

### To Reproduce Bug

1) set up 100 pixels in LED prefs
2) configure an 8x8 matrix on 2D settings page
3) add an additional segment and set start X to 64 or higher
4) (the UI will revert to a 1D segment view) enter 100 for the stop X and save.
5) After save, segment will be 2D.

### Expected Behavior

Segment beyond 2d bounds would remain 1d.

### Install Method

From srg74 firmware repository

### What version/release of MM WLED?

WLEDMM_0.14.0-b27.31_universal_shield_esp32_4MB_M

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-10-10:
Yes mixing 2D/1D does not work in MM as you have noticed  , i tested that when you reported it a while back and also now on latest MM nightly build . 
[image]

softhack007, 2023-10-10:
@ewoudwijma can you take a look?
Maybe something in the MM specific code force-resets segments in a 2D setup to 2D mode?

ewoudwijma, 2023-10-10:
Yeah, will look to it, in a few days.
I never understood the use case for this, why you want a mix of 2D and 1D? 
Why not 2 esps, one for the matrix and one for the strip ?

scottrbailey, 2023-10-10:
Well in my particular use case, the lights are in my kitchen.  I've got one
window with a strip around it and above it a semi-circular window with a
matrix in it. I want to keep out of my wife's way and use as few sockets
and as little space as possible.  But I could envision a room where you had
several windows, each with a matrix in them and a strip running between
each window.

ewowi, 2023-10-13:
I could understand that. My personal feeling is that WLED is becoming like Android, supporting more and more use cases, and I see as developer it is becoming more difficult to maintain. I think we should move to an Apple approach, support the 80% of use cases only. Of course this is a personal opinion and it is debatable what belongs to the 80% 🙂

I am quite busy and I cannot realize my earlier comment to look into it in a few days. But have it on my list , need to find time to setup a matrix and a strip configuration.

if somebody could do a code compare if this part of functionality that would help

dosipod, 2023-10-14:
I have never mentioned that before because there is a workaround  we used for a long time which  is just to setup the strip as another matrix , not optimal but works if the 2nd esp is not an option but  scottrbailey has ton of them . 
One thing that might be others do not realize and  is that 2D in MM is  a bit different then upstream which might  be toward large complex fixtures which not a bad thing . I personally do not have strips in use at all  but though unless it would just say it is not supporting then it is on the list

Finndersen, 2023-12-16:
I've managed to get 2D Matrix + 1D strip working in MM. After saving the strip config, even though it reverts to 2D mode, if you reboot WLED it should now pick up the config properly. This screenshot is from MM and works fine:
[image]

However, the issue I'm having is that I'm unable to change the 1D strip configuration to split it up into multiple segments, which I would like to do. When I try reduce the length of the segment it bugs out to 2D mode, and can't add another 1D segment

Finndersen, 2023-12-16:
Wouldn't it make sense to have 1D/2D configuration be per-segment not globally?

softhack007, 2023-12-16:
In principle, your right. 

Upstream has this whacky "mixed mode" that basicially treats LEDs "after" the matrix as a strip. Well that's also more a workaround than a solution.

The original idea was that "1D" is included into 2D, as a matrix  Nx1 is 1D. You can set any segment to mapping mode "pixels",  which treats them like 1D (not exactly, because serpentine is still applied).

softhack007, 2023-12-16:
Actually there is one technical problem with a mixed "matrix plus strip" in MM, for example you create a "72 x 1" next to a " 20x16".

Due to a very unfortunate design decision called "gap maps", wled creates a huge buffer of the combined layout, which is "92 x 16". This makes internal handling very slow.

My own workaround is to define a arbitrary matrix from the strip, say 9x8 = 72. Then use this "pseudo matrix" in mapping mode "pixels". Instead of 92x16, the buffer needed is only 29x16 pixels.

dosipod, 2023-12-16:
@Finndersen  What exact build of MM you are using to test because when i tested latest nightly MM-S  from https://wled-install.github.io/ still mixed 2d-1d mode is not working , might be take a short video /gif showing steps and end result . @scottrbailey Might be you could also retest again

Finndersen, 2023-12-17:
[image]

Finndersen, 2023-12-17:
@softhack007 so it does that even when the extra strip is properly recognised as 1D?

Finndersen, 2024-03-28:
@softhack007 @dosipod any ideas/updates on this issue? :(

dosipod, 2024-04-22:
@Finndersen Might be it will be fixed in the feature as this would also mess-up with using the esp onboard led for status led if you are using 2D setup . Does not seem to be a priority though so I personally no longer use  mixed setup on MM

Finndersen, 2024-04-22:
What does it do with the status led? I've got mixed setup working, but only with 1 strip after the 2D, can't make it with with more

dosipod, 2024-04-22:
As pics are not clear , once you able to do a video like the one here https://github.com/MoonModules/WLED/issues/79#issuecomment-1754208832 we can compare , otherwise i am not really clear on how to test that as it is still does not work for me on latest nightly build , might be visit discord and report that , status led would also be mixing 2d and 1d but that is something i could overcome in code

Finndersen, 2024-04-22:
@dosipod  do what you did in your video, then reboot WLED, as per my comment: 

> After saving the strip config, even though it reverts to 2D mode, if you reboot WLED it should now pick up the config properly.

spiro-c, 2024-04-22:
@Finndersen you need to define the matrix in 2d settings first than you can make more segments from the 1d strip and i general more 1d strips after or before 2d matrix .. how much i can se from your case you have one strip of 144 and and matrix of 13x7 so this is how it work or how @softhack007 explained 

This are led settings
[image]

Then i define two matrixes first for the strip then for the 13x7 in 2D settings
[image]

and at the end you make the segments 
[image]
 
at the end if you need more segments from the strip you can make more 2d segments but they need to be power of 12 in this case because that is how i define the first panel in 2d settings ...
[image]
[image]

Finndersen, 2024-04-22:
@spiro-c thanks I will try that, I think previously I didn't do the LED panel layout config, just tried to make segments and that didn't work. Can the LED panel layout config vary between presets?

spiro-c, 2024-04-22:
> @spiro-c thanks I will try that, I think previously I didn't do the LED panel layout config, just tried to make segments and that didn't work. Can the LED panel layout config vary between presets? As in, is it included in the preset config data?

No but you can define multiple panels from one strip with correct offset but try to make the matrix maximum dimensions as small is possible if you like better performance