MoonModules/WLED-MM #81: Keep GFX for Matrix visible while scrolling through effects

FragMenthor, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
UI problem related, GFX using matrix is hidden when scrolling through effects

[image]
[image]
[image]

**Describe the solution you'd like**
Matrix GFX Should remaing visible when scrolling though effects

**Describe alternatives you've considered**
None.

Thank YOU for making WLED better!

dosipod, 2023-10-15:
That is a very good idea which we needed as we used this to record demos but unfortunately it was not possible as @ewoudwijma  is a bit busy . The workaround for us is to open another widows with http://wled-ip/liveview2D to record the demos

FragMenthor, 2023-10-15:
> That is a very good idea which we needed as we used this to record demos but unfortunately it was not possible as @ewoudwijma is a bit busy . The workaround for us is to open another widows with http://wled-ip/liveview2D to record the demos

Yeah, but that, at least for me, is very difficult to use "as is", because of the white background.
For instace, running the "Black Hole" effect ina a white background is awfull and cannot be "understood" visually:

[image]

dosipod, 2023-10-16:
I agree with you ( even worst then you i have issue with bright lights so i must use peek ) , what i ended up doing is install another browser ( i used Firefox developer version ) and setup that with a dark background in Website appearance section 
[image]