MoonModules/WLED-MM #82: Animartrix effects: weird colors in some effects

FragMenthor, 2023-10-15:
### What happened?

Some effects give me colors that do not match the rest of the effect. Nomally is a solid green color that is supperimposed on the effect, like some limit has been reached.
Namelly:
- Complex _Kaleido_3:
[image]
- Hot_Blob:
[image]
- Lava_1:
[image]
- Module_Experiment_2:
[image]
- Module_Experiment_3:
[image]
- Module_Experiment_6:
[image]
- Module_Experiment_7:
[image]
- Module_Experiment_8:
[image]
- Module_Experiment_9:
[image]
- Parametric_Water:
[image]
- SM_10:
[image]
- Water:
[image]
- Zoom_2:
[image]

### To Reproduce Bug

Don't know if anyone more has this problem, but is shows for me in the UI and it reflects on the LEDs. This shows for me like this since the initial implementation of Animartrix in WLED.

### Expected Behavior

Smooth and continuous effects, they change abruptly using solid colors, and seem broken somehow.

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

WLEDMM_0.14.0-b27.32 esp32_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

 I don't think this is the expected normal behaviour. If this is intentional, the please close the issue. Thank you!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-10-15:
The weird colors has been noticed and reported since the initial implementation of Animartrix .as  palette is not used we thought that it is still WIP . You have described it better then we did

FragMenthor, 2023-10-15:
What is the possibility of adding pallettes to this effects?

dosipod, 2023-10-15:
@netmindz  could say as he is the one who ported the  Animartrix

netmindz, 2023-10-15:
So it doesn't really make sense to use a palette due to the way the effects are designed so I don't see Stefan ever adding that to his code.

There are a few bugs where the colours don't reset properly when you move to certain effects.

Due to the lack of a gamma correction a few don't work in quite the same way as on Smart matrix.

As for blocks of colour with no fade, I've not seen this behaviour

dosipod, 2023-10-15:
This FX192 Water example  , it is really very good as i  recall i seen it from Stefan , other then the strange superimposed  color which spoils it , thank you for the work though i do hope it could be imported even if as a future plan since you already did the hard part 
[image]

netmindz, 2023-10-17:
This fix should certainly help https://github.com/netmindz/animartrix/commit/8fd0df3e0b006244d53eaf480c2720daa5a697aa

Pulled in with https://github.com/MoonModules/WLED/pull/83

dosipod, 2023-10-17:
I have tried to experiment  and change all the values under led preference ,Buffer , Gamma ,FPS ..etc but it is worth noting that the issue is not with all effects but only few of them ( e.g waves  seems perfect) ., the one listed by the OP is somewhat still the same , FX192 Water is also still the same .  
[image]

netmindz, 2023-10-18:
Please test the patterns and report any bugs with them against

https://github.com/netmindz/animartrix/tree/OO

WLED just wraps this library. When I raised the issue of the colours of Water Stefan said it related to fact it was designed for SmartMatrix which has gamma correction

netmindz, 2023-10-21:
Quick (untested) attempt at adding gamma correction https://github.com/MoonModules/WLED/pull/85

netmindz, 2023-10-21:
Found what's wrong with water - the red is going negative and so flipping to full brightness

netmindz, 2023-10-21:
Next daily build has fixed version of rgb_sanity_check

netmindz, 2023-10-21:
Should all now be resolved I think @FragMenthor if you try this build https://github.com/MoonModules/WLED/actions/runs/6599717911 (scroll to the bottom for the assets)

FragMenthor, 2023-10-21:
> Should all now be resolved I think @FragMenthor if you try this build https://github.com/MoonModules/WLED/actions/runs/6599717911 (scroll to the bottom for the assets)

Thanks!!! I will try tomorrow as soon as I can! 😀

dosipod, 2023-10-22:
Yeah i have  tested all the effects and no more superimposed color so issue is fixed

closed 2023-10-22 by netmindz

FragMenthor, 2023-10-22:
> > Should all now be resolved I think @FragMenthor if you try this build https://github.com/MoonModules/WLED/actions/runs/6599717911 (scroll to the bottom for the assets)
> 
> Thanks!!! I will try tomorrow as soon as I can! 😀

> Should all now be resolved I think @FragMenthor if you try this build https://github.com/MoonModules/WLED/actions/runs/6599717911 (scroll to the bottom for the assets)

Thank you! All effects seem OK now! Great job! 👍 💯