MoonModules/WLED-MM #91: Cannot rotate through Playlists via Preset. No issue rotating through other Presets via Preset

IMAsIeep, 2023-12-08:
### What happened?

Trying to implement tactical button where short press with rotate through Presets (1-42) and long press will rotate through Playlists (100-105). I can set a Preset to rotate through other Presets, but cannot set a Preset that will rotate through other Playlists. Every time I either press the button to activate the Preset or manually click the Preset, it only selects the first Playlist(100) and will not move to the next one in the sequence (101/102/103...).  Note: works without issue in the AC firmware.

### To Reproduce Bug

Create Playlists with IDs 100/101/102/103 (or any other IDs)
Create Preset with API command:  P1=100&P2=103&PL=~
Each time you click Preset it selects/reselects Playlist 100. It will not move to Playlist 101

### Expected Behavior

Expectation is that each click will move to next Playlist in the sequence ie: 100, then 101, then 102.... 

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b27.31_esp32_4MB_S

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ewoudwijma, 2023-12-09:
Need to investigate this use case.

I recently used PL=r&P1=1&P2=23

preset json of this is

"100": {
      "win": "PL=r&P1=1&P2=23",
      "n": "Random",
      "ql": "R"
    }
    
This shows preset 1 to 23 in a random way and works in WLED MM.

softhack007, 2023-12-09:
> WLEDMM_0.14.0-b27.31_esp32_4MB_S
> Note: works without issue in the AC firmware

Could you try with a build from our latest source code?
b26.31 is a few months old already. We try to follow the bugfixes from upstream, so it's very likely that our latest `mdev` already has the necessary fix.

If you cannot build from source code yourself, you can use "artefacts" from the automatic nightly build.
* open a build action, like https://github.com/MoonModules/WLED/actions/runs/7113954941
* scroll down to the section titled "artefacts"
* pick the firmware .bin you need

IMAsIeep, 2023-12-10:
Awesome! I downloaded the nightly build and it worked! 
Assuming I did something wrong, when I first tried to update the firmware via OTA, it corrupted my filesystem back to 61KB. I reflashed the bootloader, which fixed the file size, but I still had the Preset issue. I went to grab the Version number to post in my reply and still showed my same custom compiled firmware version on the settings page. I had to go to esptool to erase_flash and then rewrite the bootloader and firmware for the new build to show up. Then it worked without any issue.

closed 2023-12-10 by IMAsIeep