MoonModules/WLED-MM #92: WLEDMM_0.14.0-b27.31_esp8266_4MB_M.bin Auto Off is enabled by default

clayclai, 2023-12-10:
### What happened?

I OTA'd an earlier 0.14.0 esp8266 module with binary WLEDMM_0.14.0-b27.31_esp8266_4MB_M.bin. Build 2308250 Everything worked as expect after the firmware upgrade—for 30 seconds. Then the LEDs went off. After a good night sleep and a bit of troubleshooting, I determined that the problem was that in the UI /settings/Battery->Auto Off is enabled by default Interval is also set to 30000 ms, which exclaims the 30 sec. This was confirmed on 3 copies of the same esp8266 dev. bd. No other binaries have been investigated. 

### To Reproduce Bug

Install WLEDMM_0.14.0-b27.31_esp8266_4MB_M.bin on a ESP8266EX via OTA or esptool.

### Expected Behavior

Expect WLEDMM to have start up defaults that turn the LEDs on, and leaves them on. If having the Battery->Interval set to 30000, and Auto Off enabled by default is a feature and not a bug, then the expected behavior is the observed behavior, and this is a feature request, and not a bug report

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM version 0.14.0-b27.31 Build 2308250

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-10:
I'll take a look at the battery usermod. Indeed it makes sense to have auto-off disabled by default, as well as starting with the usermod itself disabled on a freah install.

softhack007, 2023-12-30:
Fixed in release  b28-34

https://github.com/MoonModules/WLED/releases/tag/v0.14.0-beta.28

closed 2023-12-30 by softhack007