MoonModules/WLED-MM #102: Doesn't work on esp32-c3 mini

mitch030504, 2023-12-28:
### What happened?

I tried flashing this to my c3 mini but no matter how i flash it(tried flashing through esp wled flasher, tasmota ota, wled ota, esptool, and some others) the ap just wont show up. flashing the original through the web flasher works fine.

### To Reproduce Bug

flashing the firmware to an esp32-c3 mini

### Expected Behavior

for the WLED-AP to pop up and for a segment of leds to at least light up.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b28.34_esp32c3dev_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-28:
Looks like the same problem as in #101, just a slightly different symptom.

https://github.com/MoonModules/WLED/issues/101#issuecomment-1871095377

softhack007, 2023-12-29:
duplicate of #101

closed 2023-12-29 by softhack007