MoonModules/WLED-MM #105: Created Custom Palette Is not visible on palette list in main screen

houk001, 2024-01-03:
### What happened?

In WLED AC when create custom palette and click send current editor to slot its working but in WLED MM build 2312180 (athom music esp32 4MB M) edited palette is not showing on palette list but file palette0.json is created
even copy from AC content of palette0.json is not working, file looks good

### To Reproduce Bug

can't create custom palette

### Expected Behavior

Create custom palette :D

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b28.34_athom_music_esp32_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-01-03:
Hi, 

Not 100% sure, but I think that scanning for palette?.json only happens at startup. So you should restart your esp32 after uploading a palette file file, then it might become visible.

houk001, 2024-01-03:
Yes that's correct after reboot custom is visible. So on WLEDAC its after save and WLEDMM need reboot :D 
Also another difference: on WLED AC custom palette have color bar with set color and on WLED MM is not.
[image]
[image]

softhack007, 2024-01-03:
Yes. MM is behind AC functionality in some parts, as we are still using the "AC core" from summer. In other areas, we are ahead of them, like audio processing, ledmaps and 2D features.

houk001, 2024-01-03:
ok, its working so close :D