MoonModules/WLED-MM #106: Audio Responsive Custom Palette

houk001, 2024-01-03:
Add an option to Create palette to save it as Audio reactive and only thing to do is switch color on sound impulse so if i create palette with 6 colors switch between them from the first to the last and to first again, nothing more :D 
I think that will be nice feature to used it with many effects.

softhack007, 2024-01-03:
Thanks for this proposal. We might be able to enhance our existing feature "audio responsive palettes" with this 👍 (@netmindz )

> switch color on sound impulse

This is the only real problem. We could switch whenever the sound is above a certain level, however it would mean the palette is cycling constantly while music plays.
A better solution would be to cycle at beats (or onsets) only. This will be possible in the future, once I find some time to completely rewrite our current (very bad) beat detection code.

houk001, 2024-01-03:
That won't solve the problem?
[image]

softhack007, 2024-01-03:
> That won't solve the problem?

Sorry no, the dynamics limiter is not related to beat (or impulse) detection. It's just a method to smooth out strong peaks.

houk001, 2024-01-03:
ok, thanks for clarify :D Very nice beat detection is done in freestyler software for DMX management in module sound2light. its combination of auto gain control and auto treshold control. Works pretty well and I'm using it to control DMX fixture and WLED to :D  Auto gain control is working now so only treshold control is needed and its working as a time function that lower treshold for another beat after beat and rise it to full on beat so preventing beat one after another.
Page of that plugin
https://www.digiplay.nl/cms/index.php?page=sound-2-light