MoonModules/WLED-MM #113: Loss of segment preference when changing effects

ECampbell205, 2024-01-20:
### What happened?

Using the Athom build (unsure if others are impacted).  In segments,  if either "reverse direction" "mirror effect" are checked the checkbox clears every time you change and effect mode. 

[image]

### To Reproduce Bug

Check the box next to "reverse direction" or "mirror effect" in the segment.  Then change the effect, The check-box will reset and you need to go back every time to re-check

### Expected Behavior

Would expect the check-box to stick (stay checked)

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.0-b29.35_athom_music_esp32_4MB_M.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Thanks for taking look. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ECampbell205, 2024-01-28:
I flashed the non-Athon MM version, same problem.   Also flashed AC 14.1.. segment issue is gone, but sound reactivity is not as good as MM.

dosipod, 2024-05-14:
@ECampbell205  Go under the user interface menu  and uncheck "Use effect default parameters" 

[image]