MoonModules/WLED-MM #114: When creating a new preset, the entered name may be replaced by a timestamp; the preset contents may be replaced by the current state

Emstar, 2024-01-29:
### What happened?

While creating some presets containing JSON API calls, I noticed that some presets lost the name I entered. My name text was replaced by a timestamp. In some cases, the preset API text I entered was replaced by unwanted JSON which I think was a copy of the current state. 

https://i.imgur.com/BrNQkby.png

### To Reproduce Bug

Ensure you do not have preset #250, or are willing to overwrite it. 

* Click the "+ Preset" button
* Give it a name like Test
* Leave Quick Load Label blank
* Make sure _Checked segments, Save segment, Include brightness_ are all UNCHECKED
* Make sure Ledmap is _Unchanged_
* THEN uncheck _Use current state_, which reveals the API text field
* In the API field, enter any valid JSON such as `{"on":"t"}`
* Make sure Save to ID is set to 250
* Verify settings (https://i.imgur.com/BQ4fj8Y.png) and click Save

You now have a new preset 250 that looks and works as it should. To induce the error you now need to do some combination of using the preset and reloading the web UI. I cannot figure out a 100% reliable repro, but if you reload the page a few times and click the preset a few times, you should find that the name you entered is gone and replaced by a timestamp string. The API text you entered is likely gone and replaced by a new config. However, this does not always seem to happen when the name changes. 

This is what happened to my test preset as I was writing this bug. Notice the title and API text are totally different. 

[image]

### Expected Behavior

Presets should remain as configured. Names and API commands should not change without user input. 

### Install Method

From https://wled-install.github.io/

### What version/release of MM WLED?

WLEDMM 0.14.0-b29.35 (build 2401260) (MoonMod nightly version, 4 MB Flash, M: Audioreactive, Usermods incl Temp, AutoSave, 4-L Display, Rotary encoder)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I have only seen this but on presets #250 and #249, but I didn't even make a preset until I made #250. So maybe it's limited to 249-250... Or maybe it's the last two in the list. Not sure. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-01-29:
Hi, 

Preset #250 is usually used by the "autosave" usermod, which saves your current segments and effect settings regularly with a timestamp.

You can go into settings, autosave (UM), and uncheck "enabled". This should end the unexpected behaviour.

Emstar, 2024-01-29:
> Preset 250 is usually used by the "autosave" usermod...

AH HAH! Thank you so much.

Still new to WLED and mods lots of land mines yet to find I am sure...

closed 2024-01-29 by Emstar