MoonModules/WLED-MM #121: Line In pass-through for ESP32 Audio Kit

T-vK, 2024-02-26:
It is great to see that the ESP32 Audio Kit  V2.2 is supported. I managed to get it to work by following the pin configuration from the docs: https://mm.kno.wled.ge/soundreactive/Line-Input/#es8388

It would be nice to get audio pass-trough to work:

[image]

I read that it can be achieved with this board with basically zero latency, but it doesn't appear to be enabled by default. 

It would be a very valuable feature because this way you wouldn't need a mixer (or Y-adapter and deal with all the impedance issues).

T-vK, 2024-02-26:
Looks like all it would take is this:

i2s_read(I2S_NUM, i2s_buffer_read, I2S_READLEN, &i2s_bytes_read, 100);

  /* left channel */
  for (uint32_t i = 0; i < i2s_bytes_read / 2; i += 2)
   i2s_buffer_write[i] = i2s_buffer_read[i];

  /* right channel */
  for (uint32_t i = 1; i < i2s_bytes_read / 2; i += 2)
   i2s_buffer_write[i] = i2s_buffer_read[i];

  i2s_write(I2S_NUM, i2s_buffer_write, i2s_bytes_read, &i2s_bytes_written, 100);

Looking at this full example: ESP32-LyraT Audio Passthrough

softhack007, 2024-02-27:
See my response here https://github.com/Aircoookie/WLED/issues/3782#issuecomment-1967639501

I'm closing this enhancement proposal now, to avoid duplication and to prevent that the topic gets split into pieces  that have their own discussions in several repositories.

Btw, there is also https://github.com/atuline/WLED. Please don't copy your request info that repo, too. It's all maintained by the same people.

closed 2024-02-27 by softhack007

troyhacks, 2024-02-28:
This was determined to be a faulty AudioKit A1S board.

Analog passthru is working as expected with the user's replacement board.