MoonModules/WLED-MM #130: "Ripple" effects crash WLED controller

Her0brineCreeper, 2024-04-19:
### What happened?

When using one of the "Ripple" effects on my 1-dimensional SK6812 LED strip, my ESP32 randomly crashes after a few seconds and reboots. The amount of time before the crash seems random to me.

### To Reproduce Bug

Select one of the "Ripple" effects e.g. "Ripple" or "Ripple Rainbow" from the list after a fresh reboot. It works for a few seconds without problems, but then freezes, crashes and reboots the controller. 

### Expected Behavior

The effect should just work indefinitely without crashing. It clearly seems to work during the first few seconds.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.1-b30.36 esp32_4MB_S.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Device info:
[image]

LED settings:
[image]

Example video:
https://github.com/MoonModules/WLED/assets/56844781/fa9eacba-cc54-4a51-b02c-921191d43f08

If you require any further information feel free to contact me!
Thank you very much for your help! Keep up the awesome work!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-04-22:
@Her0brineCreeper Thanks for your report 👍 
I could reproduce the crash - it looks like a previous bugfix for Ripple has a copy&paste error 🤦 (confused `v` and `w`).

I'll let you know once we have a solution.

softhack007, 2024-04-22:
@Her0brineCreeper seems I've fixed it.
Can you try again with the source code from today?
In case you cannot build the firmware yourself, pick a firmware file from the automated build job:
* https://github.com/MoonModules/WLED/actions/runs/8787444187
* scroll down to "Artifacts"
* download the firmware for your device
* update by OTA, or by USB

[image]

[image]

Her0brineCreeper, 2024-04-22:
The fixed version of the effect has been running on my setup for a solid 15 minutes now without crashing. Therefore I call this issue fixed.

Thank you very much for the quick help! Keep up this amazing work!  👍

closed 2024-04-22 by Her0brineCreeper