MoonModules/WLED-MM #137: Wireguard usermod folder not included

jameskennemore, 2024-06-03:
### What happened?

The wireguard usermods directory doesn't exist, although wireguard is referenced in const.h. When re-adding the wireguard usermod directory and platformio_override.ini, Wireguard configuration options aren't in the Usermods config page after building. (It works in the non-MM build.)

### To Reproduce Bug

Add the wireguard usermod directory and platformio_override.ini, then build. Wireguard configuration options aren't in the Usermods config page.

### Expected Behavior

Wireguard config options should be in the Usermods config page after adding the wireguard usermod.

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

MoonModules v0.14.1-b30

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jameskennemore, 2024-06-03:
Added it back to the usermods_list.cpp and the configuration returned:
#ifdef USERMOD_WIREGUARD
  usermods.add(new WireguardUsermod());
#endif
#ifdef USERMOD_WIREGUARD
  #include "../usermods/wireguard/wireguard.h"
#endif

softhack007, 2024-06-04:
Hi, I'm not sure this usermod would work in MM.

We don't have the changes from upstream that allow accessing UI pages from behind a proxy (https://github.com/Aircoookie/WLED/pull/3238). Not sure if wireguard relies on this feature (I don't have a setup for testing).

aiakos-k, 2025-11-16:
# WireGuard Usermod Integration Guide

This guide describes all necessary changes to integrate the WireGuard usermod into WLED-MM.

## Files to Modify

### 1. `wled00/usermods_list.cpp`

#### Add Include Statement
Location: Around line 211, after other usermod includes (e.g., after `#ifdef USERMOD_AUTO_PLAYLIST`)

```cpp
#ifdef USERMOD_WIREGUARD
#include "../usermods/wireguard/wireguard.h"
#endif
```

#### Add Usermod Registration
Location: Inside `registerUsermods()` function, around line 414, before the closing brace

```cpp
#ifdef USERMOD_WIREGUARD
  usermods.add(new WireguardUsermod());
#endif
```

---

### 2. `platformio_override.ini` (or your environment configuration)

#### Add Library Dependency
Add to your `lib_deps` section:

```ini
lib_deps = ${esp32s3.lib_deps} ${common_mm.lib_deps_S}  ${common_mm.lib_deps_V4_M}
    ${common_mm.HUB75_lib_deps}
    ${common_mm.animartrix_lib_deps}
    https://github.com/kienvu58/WireGuard-ESP32-Arduino.git
```

#### Add Build Flag
Add to your `build_flags` section:

```ini
build_flags = ${common.build_flags} ${esp32s3.build_flags}
    ...
    -D USERMOD_WIREGUARD
```

---

### 3. `usermods/wireguard/wireguard.h`

#### Fix 1: Replace F() with SET_F() in appendConfigData()
Location: Lines 57-64

**Original:**
```cpp
void appendConfigData() {
    oappend(F("addInfo('WireGuard:host',1,'Server Hostname');"));
    oappend(F("addInfo('WireGuard:port',1,'Server Port');"));
    oappend(F("addInfo('WireGuard:ip',1,'Device IP');"));
    oappend(F("addInfo('WireGuard:psk',1,'Pre Shared Key (optional)');"));
    oappend(F("addInfo('WireGuard:pem',1,'Private Key');"));
    oappend(F("addInfo('WireGuard:pub',1,'Public Key');"));
    oappend(F("addInfo('WireGuard:tz',1,'POSIX timezone string');"));
}
```

**Fixed:**
```cpp
void appendConfigData() {
    oappend(SET_F("addInfo('WireGuard:host',1,'Server Hostname');"));
    oappend(SET_F("addInfo('WireGuard:port',1,'Server Port');"));
    oappend(SET_F("addInfo('WireGuard:ip',1,'Device IP');"));
    oappend(SET_F("addInfo('WireGuard:psk',1,'Pre Shared Key (optional)');"));
    oappend(SET_F("addInfo('WireGuard:pem',1,'Private Key');"));
    oappend(SET_F("addInfo('WireGuard:pub',1,'Public Key');"));
    oappend(SET_F("addInfo('WireGuard:tz',1,'POSIX timezone string');"));
}
```

**Reason:** WLED's `oappend()` expects `const char*`, but `F()` returns `const __FlashStringHelper*`. Use `SET_F()` macro instead.

---

#### Fix 2: Fix sscanf Format Warning in readFromConfig()
Location: Lines 87-90

**Original:**
```cpp
const char* ip_s = top["ip"];
uint8_t ip[4];
sscanf(ip_s, "%u.%u.%u.%u", &ip[0], &ip[1], &ip[2], &ip[3]);
local_ip = IPAddress(ip[0], ip[1], ip[2], ip[3]);
```

**Fixed:**
```cpp
const char* ip_s = top["ip"];
unsigned int ip[4];
sscanf(ip_s, "%u.%u.%u.%u", &ip[0], &ip[1], &ip[2], &ip[3]);
local_ip = IPAddress(ip[0], ip[1], ip[2], ip[3]);
```

**Reason:** The `%u` format specifier expects `unsigned int*`, not `uint8_t*`. Using temporary `unsigned int` variables prevents format warnings.

---

## Summary of Issues Fixed

1. **Type mismatch in `oappend()`**: Changed `F()` to `SET_F()` macro
2. **Format specifier warning**: Changed `uint8_t ip[4]` to `unsigned int ip[4]`
3. **Missing library dependency**: Added WireGuard-ESP32 library from GitHub
4. **Missing usermod registration**: Added include and registration in `usermods_list.cpp`

## Build Verification

After applying these changes, the project should compile successfully without errors:

```bash
platformio run --environment esp32S3_16MB_PSRAM_M_HUB75
```

## Configuration

The WireGuard usermod can be configured via the WLED web interface under the Usermod settings. No code-level configuration is required.

---
i have test it and run fine , but i use it with a Apollo M1 and this has other wiring.
Maybe it will help someone else too!

aiakos-k, 2025-11-16:
> @aiakos-k <https://github.com/aiakos-k>
> Thanks for sharing - you basicially describe how to copy a usermod from
> upstream into our usermod folder, then add it to usermods_list.cpp and
> solve some minor incompatibilities.
>
> I'm not sure what exactly wireGuard does, but it looks like a tool for
> remote access behind a firewall with reverse proxy.
>
> Did you test if the backported usermod allows you to access WLED-MM from
> outside of your home network?

Dear Frank,
i use it by WLED for many Device.

My Home Assistant run on a VM on my Root Server inside the RZ and yes i use
the Wireguard VPN to bridge the RZ Server to my local Device @ Home.
I have buy the Apollo M1 and to the same , what i do by WLED Deivce i add
the Usermod Wireguard.
After this i can use the Device in my Home Assistant.

I love it and its great.

best Regards,
Michael