MoonModules/WLED-MM #150: Sporadic reboots caused by WTD with hardware watchdog - WLEDMM_0.14.1-b32.40 esp8266_4MB_S, build 2406240

CLAM01, 2024-07-31:
### What happened?

I have two D1 Mini ESP8266s each that are "WLEDMM_0.14.1-b32.40 esp8266_4MB_S, build 2406240".

57 LEDS with WS2815 (12V) are connected.

It doesn't matter whether I receive and play effects via Audioreactive or whether I independently set the two WLEDs to "Solid".

A reboot always occurs after undefined times, but I can't yet determine where it comes from.

Cables are strictly separated between 12V, 5V and DIN, as well as relay output.

I measured the voltage, but it is stable at 5.1 volts during the failure.

All cables are firmly soldered, so I have ruled out a loose connection.

The free heap is constant between 18 and 19 kB.

Capacitor and 420 ohm resistor is also installed.

For comparison, I also have an ESP32 WROOM with the successor version WLEDMM_0.14.1-b32.41 operating as an audio reactive master. This one has no problems whatsoever.

I don't know what to do here and am looking for help.

### To Reproduce Bug

i dont know.

### Expected Behavior

Stable operation, just like when using an ESP32 WROOM

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.1-b32.40 esp8266_4MB_S, build 2406240

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-08-02:
Hi, 

Your wiring looks reasonable to be, maybe @lost-hope could give you more recommendations.

It looks like you're using gpio13 as a LED pin. Unfortunately this pin does not support hardware acceleration but its driven by 'bit-banging'. Bit banging on 8266 can lead to spurious WDT resets. Could you try one of the recommended pins (see below) and check if the problem still persists?

> You can use any other pin, but it will use the bitbang method, which is not recommended for reliability. It is best to stick to GPIO 1, 2, and if need be, 3.

https://mm.kno.wled.ge/features/multi-strip/#esp8266

CLAM01, 2024-08-03:
Thanks for the valuable information.

Yes, I used GPIO 13 because for some reason the WLED from Aircookie wouldn't come up (boot) at GPIO 2.

But now I have MoonModules because it also contains Audioreactive for the ESP8266. cool thing.

Yesterday I also changed the GPIO on one of the WLED lights from 13 to 2 and since then it has been stable without reboots.

The second is still on GPIO 13 and has sporadic reboots.

The soldering iron is already warm and I will also change it to GPIO 2 and watch everything again.

Doesn't it make sense at this point to point out in WLED that this port may not be suitable?

CLAM01, 2024-08-03:
another question.
It is normal that the relay takes off / on / off when i do a reboot over "info" tab?

softhack007, 2024-08-03:
> another question. It is normal that the relay takes off / on / off when i do a reboot over "info" tab or the watchdog reboots the WLED?

Yes, that could be normal, depending on your relay being active-high or active-low. 
During restart, the relay pin is controlled by the bootloader. I'm not sure if good old 8266 does initialize pins at boot 🤷 might be normal to have some "clickering".

softhack007, 2024-08-03:
this is a good overview of "special" pins (not WLED related):
https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

CLAM01, 2024-08-03:
Ok, on ESP32 the relay does not off/on/off on boot.
I have try anoter GPIO for relay on esp8266, but noting changed.
This is not good for the leds, the capacitor is 100% full an spikes of voltage can go direct to the led or not?

CLAM01, 2024-08-04:
OK, it's the same with the original WLED from Aircookie.
I read the same thing at Tasmota.
The original issue has been resolved here. I'll close the ticket at this point

closed 2024-08-04 by CLAM01