MoonModules/WLED-MM #153: DMX512 bugs, random output while receiving data from a semiprofessional transmitter.

SaderDragon, 2024-08-13:
### What happened?

While testing the DMX input feature on Wled MoonModules found a bug that input does not work right with semiprofessional DMX transmitters. While on simple FTDI based DMX transmitters, DMX input works without any problems. On transmitters that transmit DMX512 without much latency (no breaks between data frames), we get random data interference that generate random blinking of the WSLEDs on the output, no matter what DMX mode we select. 

### To Reproduce Bug

Connecting the DMX input to a more professional DMX512 transmitter, for example Eurolite USB-DMX512 PRO Interface MK2 or alike transmitter with low latency.

### Expected Behavior

Random blinking of LEDs on the output, even with all DMX512 channels set on 0.

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.1-b30.36_esp32_4MB_V4_S.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Can't really tell what DMX library is used for input, there is a mix of SparkFun DMX library and ESP_DMX from "Rickgg" even though in dmx.cpp I see parameters that are used in ESP_DMX library created by "Somewiesguy". That I used in some different DIY DMX fixtures without any issues.
Not much of a programmer myself, but I will try to look in the issue, any help will be appreciated. I can do hardware test with all the signal measurements if needed. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lippe-digital, 2024-12-16:
I guess this issue reflects on the todo list I found looking for virtual leds on mm knowledge base

> TODO
> I've proposed "DMX into WLED via a physical DMX wire (RS485)." This would be equivalent to "Effect" (and will likely be based on that mapping) so that you could plug WLED boards into a DMX512 cable and have it "do a thing" when you push a slider on the lighting control board/software. This feature does not currently exist, but is being worked on.

I would be happy to help if possible

maxroseGIT, 2025-03-26:
Hey there,

I also observe unexpected behaviour when using Wolfmix W1 DMX controller. I will repeat some of my findigs that I discuss on discord:

I have recently tested one of my WLED controlled LED strips with wired DMX input and a comercial Wolfmix W1 DMX controller and found that it does not work as expected.

-----

The setup is as follows:

Sparkfun DMX shield with XLR-3 cable

24 V RGB WS2811 LED strip with 13 LEDs on GPIO 4

WLEDMM_0.14.1-b32.40 esp32_4MB_V4_S.bin

DMX mode: Multi RGB

DMX start address: 1

-----

When the XLR-3 cable is attached to the WLED ESP32 device I can see that live DMX data is received. I assume that the WLED device consisting of 13 RGB LEDs requires 39 DMX channels in total. If I now define a DMX fixture in my Wolfmix W1 DMX controller like ch1 = R1, ch2 = G1, ch3 = B1, ch4 = R2 and so on till ch39 = B39, only the very first LED can be controlled. All other remaining 12 LEDs show maximum white.
That is a strange behaviour since the 39 channels belong to my WLED device that I see as a single fixture with 39 channels. But somehow it cannot be accessed that way.

While experimenting I found that when I define (in the Wolfmix DMX controller) 13 individual fixtures with each only 3 channels. I can control all individual LEDs of the LED strip via wired WLED DMX input. Despite this workaround somehow 'works', it is not the way how fixtures should be setup (it limits the way how animations will be processed and send out by the Wolfmix DMX controller, and it is not best practice to split up a single physical fixture into 'virtual sub-fixtures').

Does WLED recognize a single RGB LED as a single fixture? If so, is there another way to see all RGB LEDs as a single fixture with as many channels as there are RGB LEDs?

Any idea what could be the reason for this?

I also suspect that break-signals or DMX framrate might be connected with this problem.