MoonModules/WLED-MM #159: MQTT is not available in athom device b32

zandegran, 2024-08-24:
I have been using MQTT and it's not available anymore :)

softhack007, 2024-08-25:
hi,

we ha to disable some features because program space in flash was used at almost 100%.
> Flash: [==========]  96.5% (used 1518509 bytes from 1572864 bytes)

I've just changed the buildenv for athom a bit - removing ArtiFX, rotary encoder and dallas temperature support allows to re-enable MQTT.

you can find a firmware bin in the github build job
https://github.com/MoonModules/WLED/actions/runs/10549264058
scroll down to "Artifacts".

softhack007, 2024-09-03:
closing as there was no more feedback from the problem reporter.

Important: only the athom firmware binary was modified. All other firmware images are the same as before.

closed 2024-09-03 by softhack007

zandegran, 2024-09-14:
Thank you @softhack007 Working as expected

zandegran, 2024-09-14:
Does this solve mqtt for future updates though?

softhack007, 2024-09-14:
> Does this solve mqtt for future updates though?

Yes, as long as we don't run out of flash space.

Because I still prefer "no MQTT" over "no firmware".