MoonModules/WLED-MM #165: HD-WF2 flashing problem

Canditi890, 2024-09-15:
### What happened?

Today i try to flash [Huidu HD-WF2](https://www.hdwell.com/Product/index46.html) and got this in serial after flashing, wifi ap won't show up
```
[21:28:14]ESP-ROM:esp32s3-20210327
[21:28:14]Build:Mar 27 2021
[21:28:14]rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[21:28:14]Saved PC:0x403cd82e
[21:28:14]SPIWP:0xee
[21:28:14]mode:DIO, clock div:1
[21:28:14]load:0x3fce3808,len:0x39c
[21:28:14]load:0x403c9700,len:0x9bc
[21:28:14]load:0x403cc700,len:0x28dc
[21:28:14]SHA-256 comparison failed:
[21:28:14]Calculated: 5830b36eadda2ccdc73ac0028cd665f9a0ae0dcfc1ccf372c71f9cc6bad2ccc4
[21:28:14]Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[21:28:14]Attempting to boot anyway...
[21:28:14]entry 0x403c98c0
```

### To Reproduce Bug

Normal flash by esptool 

### Expected Behavior

I try to run Wled 

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.1-b32.40_esp32S3_4MB_PSRAM_S.bin/WLEDMM_0.14.1-b32.40_esp32S3_8MB_S.bin

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-09-17:
@netmindz any idea?

@Canditi890 did you install directly via USB? As this build is still experimental, you might have more luck when using "Build & Upload" (the right arrow icon) in VSCode.

Canditi890, 2024-09-17:
I upload by esptool and wled page, directly usb a-a and second try via rs232 ftdi adapter,
both metod successful end and after restart noting happen
So i need change bootloader too and cannot be done by esptool?

netmindz, 2024-09-17:
You are not using the correct image

https://mm.kno.wled.ge/2D/HUB75/

States you must use esp32S3_4MB_S

Canditi890, 2024-09-17:
It was precompiled? i looking through last assets and can't find that one

netmindz, 2024-09-17:
It's newer than the latest release, so you need the latest mdev build - https://github.com/MoonModules/WLED/actions/runs/10907015366

Canditi890, 2024-09-17:
Thanks, and do some tests and work fine with 128x64 p2.5 panel (set 64x64 2 chains)

Artnet doesn't support this panel because it needs 40 more universes,
but ddp works perfectly, even you can play games on it through hyperion stream.

I also found one bug, after first setting led outputs don't save any further changes, restarting won't fix it. you have to re-flash
[image]

closed 2024-09-17 by Canditi890