MoonModules/WLED-MM #172: Best way to use GIFs with HUB75 screens.

1liminal1, 2024-10-24:
**Is your feature request related to a problem? Please describe.**
I have a HUB75 64x64 screen that I'm using in a BMO. I want to create a face and then have it go to an effect after 3 seconds or so.

[image]

Ive tried using https://github.com/ajotanc/PixelMagicTool But 64x64 may not work; 

https://github.com/ajotanc/PixelMagicTool/issues/4#issuecomment-2431751858

Ive tried using https://github.com/werkstrom/WLED-PixelArtConverter/ and while this does allow me to upload frames to the device, I cant seem to save these as presets. (well I can, but they dont seem to save as different "images" though I see the different images on the HUB75 device. I posted about this issue here https://discord.com/channels/700041398778331156/1298611305002831912/1298611309029490708)

**Describe the solution you'd like**
An easy way to add pixel art to WLED with either 

Gif branch https://github.com/Aircoookie/WLED/tree/gif , the bins are here if you cant compile from source https://github.com/dosipod/WLED_gif_update/actions/runs/10116313741 , as I re-based that on 15
Usermod (client and server ) https://github.com/hughc/wled-pixelart-client

**Describe alternatives you've considered**
Stated above 

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

softhack007, 2024-10-28:
Hi, nice project 😃 

the main problem with all options is available RAM. The HUB75 driver is very hungry for RAM, however storing a 64x64 bitmap in a preset creates a huge JSON file that cannot be parsed again on classic esp32. Are you using esp32, or esp32-S3? 
On an esp32-S3 with PSRAM you might be lucky, and it could work.

As alternative solution, try to make your "face" segment smaller than the panel - like 32x32 or maybe 24x24. For a smaller segment, it should be possible to show a pixel graphic. 

Or - in case you don't mind a "blocky" look - you can set the segment to "grouping: 2" (plus "spacing: 1" for real blocks) and keep it at full 64x64 size.

softhack007, 2024-10-28:
@1liminal1 @dosipod I've played with images a bit. 
WLEDMM needed a small patch to display images from preset --> https://github.com/MoonModules/WLED/commit/f14c4cbe5b42974237a6b5c7de1f4afc255029fc

Now it works the same way as in upstream.

Using [PixelArtConverter](https://github.com/werkstrom/WLED-PixelArtConverter/), i was able to upload images to my HUB75 test setup (esp32-S3 with PSRAM, 128x64 HUB75 panel). 

[image]

<br/> 

Copy & Paste for creating a 64x64 preset first did not work, but I managed with a bit of manual "post-processing". Basicially you need to join all the sections manually, and join the array segments into one array --> use one `,` to replace `"]}}
<NEXT COMMAND (multiple commands not supported in API/preset setup)>
{"on":true,"bri":128,"seg":{"id":0,"i":[`

If you want to test something, copy everything from the attached file into new preset in WLEDMM:
[preset64x64_donkey.json](https://github.com/user-attachments/files/17542862/preset64x64_donkey.json)

For the future, we could look into adopting the "GIF" branch from Aircoookie. 
Plus there is a nice API from espressif for loading JPG files, which we could use to add a "static background" for effects.

dosipod, 2024-10-28:
@softhack007  Excellent and I will surely test that  . 
I would just like to mention that  PixelArtConverter  was not really meant to be used with gifs  but rather with pics and I was closely involved with the owner during initial stage and sadly he is not having the time to improve it  when it comes to gif and the use of playlist and multiple presets is a door to a lot of issues even if  that is used with small matrix on a normal esp32 .
Adopting the "GIF" branch from AC might be something to focus on long term as it is truly better when we tested and I would really love to see that on MM or HUB75 is added to AC . The usermod is also very much a good option which I personally use and solved the space issue but might be the standalone option is what is practical at this time for the masses  .
Thank you .

1liminal1, 2024-10-28:
> Copy & Paste for creating a 64x64 preset first did not work, but I managed with a bit of manual "post-processing". Basicially you need to join all the sections manually; i.e. join the array segments into one array --> use one , to replace ]}} <NEXT COMMAND (multiple commands not supported in API/preset setup)> {"on":true,"bri":128,"seg":{"id":0,"i":[

Thanks so much for that :) I thought that might be the case and tried joining the parts last week, but I guess I need that update? Is that now in the nightly for MM? (LOL, Im prob getting the lingo wrong, I come from a systems background) Let me know and I'll update. 

Im guessing I need to wait till its the 29th? I'm not really sure how these things work; 

https://wled-install.github.io/

[image]

Ill test the DK json when I get home and let you know if that works. If I can get three frames of BMOs face in a playlist ill be happy :) 

The plan is that a mmwave will turn him on when I walk in the room, it's his face for 3s and then goes to soap or something :) I think it will be cool. Ill be posting the remix for the STLs at some point, if you love BMO and want to make him, I'll post the link at some point :)

1liminal1, 2024-10-28:
> The usermod is also very much a good option which I personally use and solved the space issue but might be the standalone option is what is practical at this time for the masses .

Ok, so you're saying if I use the https://github.com/hughc/wled-pixelart-client this as a usermod and compile my own binary, it will work? 

Sorry to be a pain, I have compiled stuff before, but just to be clear, so I download the MM code base(?) and then following these instructions 

https://mm.kno.wled.ge/advanced/compiling-wled/

Set up VSCode and PlatformIO ect and then following the instructions on wled-pixelart-client to get the usermod to add in the usermod. And this will have hub75. Or I need to add that as well?

1liminal1, 2024-10-28:
Oh, and last comment lol. Im using a Trinity ESP32

dosipod, 2024-10-28:
For the usermod ,you would have to add to MM first , i have added it to AC here https://github.com/dosipod/WLED_15/tree/ALDIY_15/usermods/pixelart_client  and then try it with MM HUB75 bin which i never did .

I am not really sure of how stable HUB75 on a esp32 as I faced issue with that a while back so I got  HD-WF2 and it works fine ( this is a cheap  s3 board ) 
https://www.aliexpress.com/item/1005002258734810.html
You might also want to take a peek at what is done with HUB75 panels in MM discord  https://discord.gg/JxHzbzu5 as I am a bit out of touch with that but will for sure do a lot of hardcore testing around December  as I will be getting more HW and guys to help in testing if all goes to plan 

Cheers

1liminal1, 2024-10-29:
Awesome :) thanks for that. 

Im happy to help testing as well. I have a few more HUB75 64x64 and Trintys coming. I want to see if I can string 3 64x64s together and make one of my Prisms out of it. WHen you see some of the things Troy is doing with HUB75 on the S3 with 128x128... Its just amazing!

dosipod, 2024-10-29:
That is good you are aware of what is going on , since you are aware of what Troy is doing ,might be you are on discord already and I think that would be the place to help for testing but not really sure about using more "Trintys" as it seems to be too much for esp32 vs s3 board I listed if  the performance okay for you then why not

1liminal1, 2024-10-29:
HAHA, we have spoken a couple of times on Discord :) You and Wildcats helped me with Pixel Mapping for my prisms. 

[image]

I actually posted in support and the standard WLED discord about this

[image]

But its easy for things to get lost on Discord, people are busy with their own projects ect ect. SO thanks for taking the time to give me some advice, I appreciate it :) 

https://esp32trinity.com/

This is the Trinity, its working well actually with any of the default effects. Not sure how it will scale, but will see.

softhack007, 2024-10-29:
> This is the Trinity, its working well actually with any of the default effects. Not sure how it will scale, but will see.

Hi @1liminal1, the trinity is based on "classic" esp32, right? 
Esp32 works well up to 64x64, but it gets unstable if you have a larger setup with more panels.

For more than 64x64 pixels HUB75, you should use *esp32-S3* - preferably with octal PSRAM.

See https://mm.kno.wled.ge/2D/HUB75/#panel-size-and-chain-length

dosipod, 2024-10-29:
I have only tested MM briefly on a normal esp32 I was using with  https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA   and in that I faced few issues but might be that is fixed or something about trinity that is different . It is still the case that the guys recommend s3  hence I got  HD-WF2 but I also only plan to use single 64x64 per esp so might be esp32 is okay . And I thought the avatar looked familiar :)

1liminal1, 2024-10-31:
Apologies guys, I haven't had time to test the JSON or do any more testing with the Trinity. 

Too many projects!!! LOL been working on this Retro Pi BMO :) 

[image]

But hopefully ill get to it tomorrow. 

Thanks again for taking the time to help me with this :)

dosipod, 2024-11-01:
I normally  do not like to send others on a wild guess chase  so I was trying to test each option and the first one is the usermod which  I have now added to MM but the size was an issue so i had to kill animatrix  for it to fit  on the   HD-WF2 s3 board  

[image]
The bin i used is the env esp32S3_4MB_S_ALDIY in here 
https://github.com/dosipod/MM_OCT_29/blob/mdev/platformio.ini

After that some of the pics worked okayish ( some crashes are there ) but gifs never worked at all on  HD-WF2  , i have a node.js local server somewhere which i could try also but things do not look so promising . The full  video of the test is too large to fit here so I had to reduce the size 
[image]

dosipod, 2024-11-01:
@softhack007 I will try all you listed as I have setup node.js locally now but seeing a bit of odd behavior when testing on latest AC which may have to do with the gif sizes ( the same gifs  worked fine on AC gif branch )  and once i sort that out I will for sure test a lot on MM  , I am still hoping that might be with the change of license that we could focus on the end goal ( either gif branch on MM or HUB75 on ac ) as this is just a temporary option but will try our best .

1liminal1, 2024-11-02:
Hey guys, Just going though the posts now. 

I have ordered some HD-WF2, wish I had of known about those before I ordered the Trinitys, far far cheaper :( lol, Anyway. 

> If you want to test something, copy everything from the attached file into a new preset in WLEDMM:
> [preset64x64_donkey.json](https://github.com/user-attachments/files/17542862/preset64x64_donkey.json)
> It should display our king from the 1980's.

So, just to be clear, I copy the json and put it into a preset yes? 
I get this error. 

[image]

> I am not really sure of how stable HUB75 on a esp32 as I faced issue with that a while back so I got HD-WF2 and it works fine ( this is a cheap s3 board )

I missed that before :( lol, wish I had of seen that before ordering more Trinities LOL

So I guess the Trinities are a dead end lol

OK, so I just tried merging my own file and it only reads the first few lines. I guess that's why pixel art splits them in the first place. 

So I tried making a playlist out of that and it does work; 

https://github.com/user-attachments/assets/b30dd36f-3f2a-402b-a883-41ca72e76fa5

But the issue is I added soap to the end and when It tries to play that it just goes turns off (I see the power button on the webUI go off). 

[image]

Also I cant set the playlist lengthy for anything less then 1 sec, so its super janky anyway lol

dosipod, 2024-11-05:
@softhack007 @1liminal1  A stable workaround found , I used the gif branch  ( only changed MAX_LEDS_PER_BUS 4096 in const.h so I could use DDP with 64x64 running on  MM ) 
https://github.com/dosipod/WLED_gif_update/tree/gif_64x64
https://github.com/dosipod/WLED_gif_update/actions/runs/11676697887  ( use firmware-ALDIY_esp32dev_audioreactive_gif.bin ) 
And behold gifs are working on 64x64 without any issue  . 
The below video is from a classic esp32 running AC 15.0.b4 gif branch and controlling MM build with hub75 64x64
[image]

Might be we could see value in adding gifs to MM else this will have to do for now

1liminal1, 2024-11-07:
Hey, so I updated to use firmware-ALDIY_esp32dev_audioreactive_gif.bin 

But there is no hub75, you mentioned DDP  but that needs as IP ect. What am I doing wrong here. 

And how do I get the gifs on there. Is it this? 

https://github.com/hughc/pixel-art-server

dosipod, 2024-11-07:
@1liminal1  
Yes you would use firmware-ALDIY_esp32dev_audioreactive_gif.bin on a classic esp32  as DDP master running gif branch  . 
The ip would be for your other unit running MM with hub75 support 

[image]

1liminal1, 2024-11-07:
Ahh ok, so I need to run two, I just updated my hub75 ESP HAHA. 

Ok cool, no worries. I have plenty of ESPs around lol

Ill let you know how I go :) Thanks mate! I appreciate it

dosipod, 2024-11-07:
This is just a workaround as we can not wait for gif to be added to MM or for hub75 to be added to AC so we could use one esp .
There is another advantage to using a 2nd esp which you will notice later if you use a lot of gifs that you would run out of space . SD card support is there but at the moment not sure it is possible to use that with gifs . If you have esp32 with 8 or 16 MB flash then might be best to use that .

1liminal1, 2024-11-08:
So been testing it for a bit and its pretty much unusable on a ESP32. Pages are not loading, 

Got the gif, uploaded
I have it set as ddp (pretty much like your screen shot) and 2d matrix 64x64 
Named the segment the same as the gif and also named the preset the same as well?

And then is just stopped responding. Power cycle, makes no difference  

Have you run this on a ESP32 or just the S3. 

I have an ESP32-S3-WROOM-1 N8R2 N16R8 44Pin Type-C 8M PSRAM 

If you have a bin I can try?

dosipod, 2024-11-08:
For the gif branch DDP master I am using classic esp32 and for the Hub75 I am using  HD-WF2 

All the MM bins last CI run is here , 
https://github.com/MoonModules/WLED/actions/runs/11730438023
( I do not have S3 16MB flash to test with Hub75 )

softhack007, 2024-11-08:
> ( I do not have S3 16MB flash to test with Hub75 )

You can use `esp32S3_4MB_S` on the HD-WF2; 
T7-S3 16MB flash would be best as it has PSRAM, so more free RAM to play with pixel graphics....

dosipod, 2024-11-08:
I would love to get T7-S3 personally as I do not look forward to doing any manual wiring as it was a pain before  but can only afford to get below board for now  
[image]

1liminal1, 2024-11-11:
So my trinity responds ok, and if I go to the web server, I actually get the streaming dialogue for DDP. But nothing happens. 

Is there a limit of gif size? (I'm at work and cant remember lol, it was either 15kb or 30kb.)

So just to be clear, 

EPS32 Trinity for the Hub75 (with some HD-WF2 on the way)

And a ESP32dev board for the gif branch. Which is pretty much not responding, Ill see if I can get to the setting page with the ram usage and let post it here. 

Also it was a fresh install of AC (thats the base version right?) from the webtools and then I updated to your Bin Frank :)

dosipod, 2024-11-11:
This is the gif files I tested with. So I think the easy way is for you to test with the gif branch bin and ddp to any other  smaller unit first and then test on the hub75 after  . I am using HD-WF2 with MM for the hub as I did face memory issue when I tested using classic esp32 but I thought since that was a while back it might have been better by now .
To be honest the HD-WF2 is very easy to flash and now  stable and I only had to serial flash it one time two weeks ago and since I have updated it with OTA many times and now not facing any real issue with MM  and I am testing it a lot but will be getting the s3 above just to test multiple panels as softhack advised but I do not really plan to use more then one panel  per setup 

[Gif_branch.zip](https://github.com/user-attachments/files/17694375/Gif_branch.zip)

8bignic8, 2024-11-25:
I´m trying to connect the hub75 port to an esp32s3 does anyone know how to connect the pins? Because in the official wiki and the pinout for the esp32 there are pins that the wroom esp32s3 does not have :D. So if somone could give me a hint that would be amazing.

dosipod, 2024-11-25:
@8bignic8  Good question  , there is an example of s3 wiring here 
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA 
If that works for you the please link to your board

softhack007, 2024-11-25:
> I´m trying to connect the hub75 port to an esp32s3 does anyone know how to connect the pins? Because in the official wiki and the pinout for the esp32 there are pins that the wroom esp32s3 does not have :D. So if somone could give me a hint that would be amazing.

Hi,
Current you only find this information in the source code.

Which build do you use? 

* `esp32S3_16MB_PSRAM_M_HUB75` uses the "MoonHub" pinout: 
   https://github.com/MoonModules/WLED/blob/7d10bd0cd99982b4e2332d687c9f19f4e8dc566b/wled00/bus_manager.cpp#L751-L752

* `adafruit_matrixportal_esp32s3`
   https://github.com/MoonModules/WLED/blob/7d10bd0cd99982b4e2332d687c9f19f4e8dc566b/wled00/bus_manager.cpp#L727

* `esp32S3_4MB_S` (for HD-WF2), and `esp32S3_8MB_M`: 
   https://github.com/MoonModules/WLED/blob/7d10bd0cd99982b4e2332d687c9f19f4e8dc566b/wled00/bus_manager.cpp#L781
   
* `esp32S3_8MB_PSRAM_M` and `esp32S3_WROOM-2_M`
https://github.com/MoonModules/WLED/blob/7d10bd0cd99982b4e2332d687c9f19f4e8dc566b/wled00/bus_manager.cpp#L755

8bignic8, 2024-11-26:
Im currently using: WLEDMM_0.14.1-b32.41.dev_esp32S3_16MB_PSRAM_M_HUB75-> i dont have the pinout for it yet so i dont kow if it will work. 
I could not find the .bin for esp32S3_WROOM-2_M, 

Im using a wroom 32S3-1 16MB, the adafruit one was unstable and could not allocate enough ram for my 64x128px matrix.

What worked to flash the software was to first flash the tasmota-4mb version than use the ota flasher after connecting to the tasmota ui of the esp32S3, and afterwards selecting and uploading the choosen wledxxxxxx.bin file :)

softhack007, 2024-11-26:
> first flash the tasmota-4mb version than use the ota flasher

Well, this works exactly once. 
The OTA partition from tasmota is too small, so OTA updating WLED with a newer build will fail. OTA update cannot replace the partition file, so next update you do must be via USB, See https://github.com/Aircoookie/WLED/issues/4241.

> I could not find the .bin for esp32S3_WROOM-2_M

If the normal "esp32S3_16MB_PSRAM_M" firmware works for you, it means you don't have WROOM-2. The reason we added the  WROOM-2 buildenv is that these boards need their own special firmare, and they won't boot with anything else.

> currently using: WLEDMM_0.14.1-b32.41.dev_esp32S3_16MB_PSRAM_M_HUB75

This one uses the "Moonhub" pinout. See the first bullet in my [previous answer](https://github.com/MoonModules/WLED/issues/172#issuecomment-2499115440)
(does anybody read what I'm writing 😜 ?)

dosipod, 2024-11-26:
@softhack007 Read yes but understand not all the time  :) , I will try to make a picture of the wiring if I get s3 that I could wire and test with HUB75 but I remember a long time ago that on esp32    with DMA project that you could use any pins as long as it matches what you assigned in the sketch so the example you listed from bus_manger.cpp is fine

1liminal1, 2024-11-26:
Yeah, ill second that. A lot of times I have no context for what you guys are saying, so I read it. But I make a wrong assumption about what it means HAHA. We are all learning though :) I know you guys have helped me a lot :)

8bignic8, 2024-11-29:
> > first flash the tasmota-4mb version than use the ota flasher
> 
> Well, this works exactly once. The OTA partition from tasmota is too small, so OTA updating WLED with a newer build will fail. OTA update cannot replace the partition file, so next update you do must be via USB, See [Aircoookie#4241](https://github.com/Aircoookie/WLED/issues/4241).
> 
> > I could not find the .bin for esp32S3_WROOM-2_M
> 
> If the normal "esp32S3_16MB_PSRAM_M" firmware works for you, it means you don't have WROOM-2. The reason we added the WROOM-2 buildenv is that these boards need their own special firmware, and they won't boot with anything else.
> 
> > currently using: WLEDMM_0.14.1-b32.41.dev_esp32S3_16MB_PSRAM_M_HUB75
> 
> This one uses the "Moonhub" pinout. See the first bullet in my [previous answer](https://github.com/MoonModules/WLED/issues/172#issuecomment-2499115440) (does anybody read what I'm writing 😜 ?)

hey sorry yes and amazing thanks to that. Today I will try: 
| HUB57  | ESP32S3 16MB |
|-----------|--------------|
| R1_PIN    | 1            |
| G1_PIN    | 5            |
| B1_PIN    | 6            |
| R2_PIN    | 7            |
| G2_PIN    | 13           |
| B2_PIN    | 9            |
| A_PIN     | 16           |
| B_PIN     | 48           |
| C_PIN     | 47           |
| D_PIN     | 21           |
| E_PIN     | 38           |
| LAT_PIN   | 8            |
| OE_PIN    | 4            |
| CLK_PIN   | 18           |

The Setup with the adafruit_matrixportal_esp32s3 and 64x64 matrix and lenght 2 resulted in a freezed esp32/ui because of a memory problem. only one 64px mirrowd image was possible.

softhack007, 2024-11-29:
> resulted in a frozen esp32/ui because of a memory problem. only one 64px mirrored image was possible.

Yeah it depends on the effect you want to run - WLED wasn't designed as a screensaver engine, so bigger displays won't work well with some effects. It is as it is.

Especially animArtix effects tend to become slow, and some won't work above 48x48 due to memory needed by the effect.

softhack007, 2024-11-29:
> Setup with the adafruit_matrixportal_esp32s3

The adafruit matrixportal firmware is exactly for that board and nothing else.
You cannot freely "juggle" with different S3 binaries. If `esp32S3_16MB_PSRAM_M_HUB75` works for you, then keep using it.

dosipod, 2024-11-29:
My HD-WF2 with 64x64 is  slow  specially when streaming  , so yeah I will be using  grouping , more buffer and websocket disabled . 
When I tried to stream it only kinda worked from xligts with ddp ( very very slow and not all effects worked  ) , Hyperion and HyperHRD are not supporting 64x64 yet ( I raised a case for that and they are making a fix ). 

I also tried from Jinx with art-net and sACN and could only stream to half of it  so might be it is only local effects for now on that HD-WF2 board . Not sure there is an affordable s3 Ethernet board  might be  a bit better but Troy warned about wifi so it was expected but I thought it would not be much of an issue on 64x64  but apparently it is

softhack007, 2024-11-29:
> My HD-WF2 with 64x64 is slow specially when streaming

Could be true, as this "path" for setting pixels did not receive much attention and optimization yet. 
We might be able to look into real time streaming - and search for bottlenecks - next year. For now (=upcoming MM release) the focus was on build-in local effects.

The bottleneck could be network receiving, to small buffers, or very inefficient handling inside `setRealtimePixel()` - Or all of them. We'll find out next year.

8bignic8, 2024-11-29:
from my side total love to everyone that works on a wled distro and/or feature its amazing work you all doing tanks:)!

netmindz, 2025-09-20:
WLED-MM now includes GIF playback using the "Image" effect

closed 2025-09-20 by netmindz