MoonModules/WLED-MM #173: esp32c3dev_4MB_M is too big to fit

dosipod, 2024-10-24:
Using https://wled-install.github.io/ to flash esp32c3dev_4MB_M does not work due to size and OTA will not work .
On AC the build works  as  in this example 
https://github.com/dosipod/PUB-WLD-GH-COMILE/actions/runs/11502411179
But only after we disabled a lot of features which we also disabled in MM but still it does not work .
This was reported in MM discord a while back but might be discord needs a CPR .

I also  tried to gz the bin but does not work  , if c3  issues are of interest then we could do more testing else I hate c3 also

softhack007, 2024-10-27:
Hi, thanks for your report 👍 

> Using https://wled-install.github.io/ to flash esp32c3dev_4MB_M does not work due to size and OTA will not work .

the current statistics are
> Flash: [==========]  99.8% (used 1569990 bytes from 1572864 bytes)

maybe this is already too big for OTA 🤔 
I've just removed ARTI-FX, which is a pity, but it brings us down to 97.5% Flash usage. The next option would be to remove IR support.

Please try again with the latest build artefacts: https://github.com/MoonModules/WLED/actions/runs/11543379034

I think the ultimate solution - for all 4MB boards - is we move to the 512Kb LittleFS partition table. this will add ~256kb of extra program space. 

> if c3 issues are of interest then we could do more testing else I hate c3 also

Same here 😉 Its like 4 months ago that I used a C3 board. C3 is very low-end for the advanced stuff we try in MM ... However good to know that we can still provide something for users to play with - for a start before buying a better -S3.

dosipod, 2024-10-28:
@softhack007 Thank you for the reply and looking at this .
Unfortunately not only OTA fails but also serial  flashing  with the above bin , after flashing there is a boot loop in MM.
Beside the size which 1541792 bytes in AC vs 1650960 bytes in MM , I see that Flash Mode  is dout in AC but dio in MM.
I have attached two text files marked AC and MM  which has the env used in AC  and serial log from both and the boot loop in MM .

I do not wish that you invest any more time on this ( Although having none-working  bin on the installer site is bad ) as I will do all the testing and changes to see what fits with my little knowledge of MM build environment as of now .Not looking forward  to remove things like animatrix as that is one of features I wanted to show but we will still try to make c3 work in any way possible with MM even though personally I do not think the $2 price tag  is worth the effort as I get esp32 for $3 but some of the guys are promoting c3 and hence other users will buy it not knowing this kind of issues nor that you as  the main wled dev  do not recommend using it  . Have a nice day 
 
[ESP32-C3DEV-AC-15.0.B6_USB_FLASHING.txt](https://github.com/user-attachments/files/17535823/ESP32-C3DEV-AC-15.0.B6_USB_FLASHING.txt)

[ESP32-C3DEV-MM_281024_USB_FLASHING.txt](https://github.com/user-attachments/files/17535824/ESP32-C3DEV-MM_281024_USB_FLASHING.txt)

dosipod, 2024-10-28:
I just missed there is two bins on the link so I will try with firmware-esp32c3mini_dio_4MB_M  , wait for the kill shot a bit

dosipod, 2024-10-30:
After some tinkering we have came to the conclusion that C3 board does not fit our use cases if AR is needed so we will no longer use it or test with AR ,this is not limited to MM but also in AC as we faced a lot of issues with it  when AR is enabled and performance was sub-optimal. 
@softhack007 You may close at your discretion and thank you for the help .

softhack007, 2024-11-01:
> some of the guys are promoting c3 and hence other users will buy it not knowing this kind of issues

@dosipod just for my curiosity - one the "the guys" may be known as "the frog" ?

dosipod, 2024-11-01:
@softhack007 Unfortunately there are many who do that  without knowing  other guys will go and buy them  . I was one of the guys who did not listen to you a long time when it comes to AR and C3 . It is purely lack of understanding of wled and the board itself  . This particular  c3 suppermini is of concern due to possible design issues, poor heat dissipation  and wifi issues which I did not longer test  much as at the end even if it works I do not really have a use case for low end esp  but now we  rather focus on S3 that could bring value to wled .