MoonModules/WLED-MM #178: Soundsync is Nott working over esp32 own AP

up2smoke, 2024-11-05:
### What happened?

Hello guys. I have the following question: I use wLED moonmodul. In particular because of the sound sync function. I use an ESP32 with inmp441 as the transmitter and an esp8266 as the receiver. Everything works so far. However, I also want the esp32 to create a WiFi AP that the esp8266 can log into. This is exactly the point where the sound sync function no longer works. It seems as if the sound sync function is not being sent if the esp32 itself is the access point. Does anyone know of a solution?

### To Reproduce Bug

Activate Soundreactive and the transmitter/receiver function. Connect the transmitter (esp32) and receiver (esp8266) to your WiFi - everything works.
create a WiFi on the ESP32 so that the esp8266 can connect to it - Soundsync no longer works

### Expected Behavior

I thought it makes no difference if the esp32 is in a WiFi, or if the esp32 itself is the wifi

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

0.14.1-b32 / esp32 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-11-05:
Audio sync also does not work on AC if the slave unit is not connected to the same wifi . I thought that is expected as in the audio sync the port  for send/receive might not be valid if we are in AP mode . But will be nice if that's works .

I recall there was work done in MM for audio sync using ESPNOW but not sure of status .

dosipod, 2024-11-16:
I was looking at the audio usermod for other thing and I see the notes about AP mode so might be it was supposed to work 
[image]