MoonModules/WLED-MM #180: lolin_s2_mini environment missing from CI run and web installer

dosipod, 2024-11-09:
### What happened?

lolin_s2_mini environment missing from CI run and web installer  https://wled-install.github.io/  , the one generated is esp32s2_PSRAM_M which will not fit the lolin_s2_mini board  so as a result the update will fail due to size as reported by other users 

Checking size .pio/build/esp32s2_PSRAM_M/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.5% (used 67256 bytes from 327680 bytes)
Flash: [========  ]  84.9% (used 1613546 bytes from 1900544 bytes)
Building .pio/build/esp32s2_PSRAM_M/firmware.bin

Checking size .pio/build/lolin_s2_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.8% (used 64832 bytes from 327680 bytes)
Flash: [========= ]  92.2% (used 1449670 bytes from 1572864 bytes)
Building .pio/build/lolin_s2_mini/firmware.bin

For installer https://wled-install.github.io/ might be @wled-install  can take a look ( I hope this is the correct username for Wladi) 

A test run with both environment is below and I only have one lolin_s2_mini so not sure what is esp32s2_PSRAM_M is for but if that is for another board then might be a good idea to indicate that 

https://github.com/dosipod/MM_9_NOV_2024/actions/runs/11758362634

Note to softhack , this might be the same case with lolin c3  which we raised before so will try that and update the other open issue 

### To Reproduce Bug

Use the bins from https://wled-install.github.io/ or CI run to update lolin_s2_mini 

### Expected Behavior

Update should be okay

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

latest 

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Anything else?

Take it easy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wled-install, 2024-11-10:
The issue is that [env:esp32s2_PSRAM_M] enviroment uses another partitions table (WLED_ESP32_4MB_256KB_FS.csv), but my installer flashes the same partitions for all devices with 4MB (default si 4MB / 1MB FS). As a result it flashes wrong partition table for S2. @softhack007 is it intended to have different table for S2 only? If yes, I would change the installer.

dosipod, 2024-11-10:
@wled-install  Thanks for clarifying that , since you are here please take a look at this https://github.com/MoonModules/WLED/issues/173 as we have faced an issue with flashing MM on c3 from installer . I think that is due to bin size as we had to drop other features in order for the bin to fit on AC with audio  but I failed to do the same for MM not sure what to drop . Otherwise just for you to be aware of that

softhack007, 2024-11-10:
Hi @wled-install, thanks for you report :-)

> The issue is that [env:esp32s2_PSRAM_M] environment uses another partitions table (WLED_ESP32_4MB_256KB_FS.csv), but my installer flashes the same partitions for all devices with 4MB (default si 4MB / 1MB FS). As a result it flashes wrong partition table for S2. @softhack007 is it intended to have different table for S2 only? If yes, I would change the installer.

In fact this is an interesting topic. Right now, all 4MB builds are either at the limits of the 4MB / 1MB FS, or above the limit.
That's why we moved (intentionally) to smaller FS sizes. Actually, this will happen soon in AC WLED, too, because more features imply larger program size.

We have several smaller FS partitions files: 256KB (WLED_ESP32_4MB_256KB_FS.csv), 512KB (WLED_ESP32_4MB_512KB_FS.csv) and 700KB(WLED_ESP32_4MB_700k_FS.csv).

I'd suggest you use the 512KB FS for all 4MB boards - or maybe the 700k_FS one if more FS space is important to your users. I think it will work to actually use the 512KB partition also on firmware that was build for 1MB FS.

For "immediate rescue", i've just added esp32s2_PSRAM_S which still fits into 1MB FS. Actually I would not recommend to use the AC "standard" buildenv instead of MM special ones. Simply because AC "standard" builds are neither tested nor used by the MM team.

cheers

dosipod, 2024-11-11:
@softhack007  Thank you ,the esp32s2_PSRAM_S environment now worked on the lolin_s2_mini with OTA . Might be change the comment you added to the platformio.ini as esp32s2_PSRAM_M does work on lolin_s2_mini

[image]

Not sure that was reflected on the installer but will test tomorrow or once @wled-install confirmed

wled-install, 2024-11-11:
@softhack007 Thank you for clarification!

@dosipod I've updated wled-install, now it automatically uses correct partitions.bin for nitghly builds (just takes the one generated by the build itself). For now only for S2 / MM, seems work well, later will check it more detailed and then do it for all targets (and also for AC nightly builds).

dosipod, 2024-11-11:
@wled-install I was trying with OTA for now but after downloading the bin that fails as the MM env listed on the site does not include the S build so only M is there . Might be I am looking at the wrong place 

[image]

wled-install, 2024-11-13:
@dosipod It was just no included in nithly builds, now it is, but will appear after next run