MoonModules/WLED-MM #183: ESPhome flasher mess with bootloader

Canditi890, 2024-11-12:
### What happened?

Somehow, when I flash using esphome flasher the batch loads in the wrong place and messes with the bootloader

A month ago everything was ok, I had to reload the file because I messed with the wifi settings

Probably this is esphome's fault, just reload the [bootloader](https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/S3_bootloader.bin)  at 0x0 and everything works

### To Reproduce Bug

[image]

### Expected Behavior

Proper flashing

### Install Method

Self-Compiled or other

### What version/release of MM WLED?

Huidu HD-WF2 ESP32-S3

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-11-12:
@Canditi890  Have you tried the one modified for wled ? https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_%20ESP_Flasher  ,  esphome flasher caused issues since a long time ago 

For s3 you may also try to flash directly from this site https://wled-install.github.io/

dosipod, 2024-11-12:
I was  trying to verify from the installer but I just noticed that the OP listed HD-WF2 which would need the 
MM env esp32S3_4MB_S but that is not available for MM on https://wled-install.github.io/  .Only ESP32-S3 8 MB
is listed for MM @wled-install

Canditi890, 2024-11-12:
> I was trying to verify from the installer but I just noticed that the OP listed HD-WF2 which would need the MM env esp32S3_4MB_S but that is not available for MM on https://wled-install.github.io/ .Only ESP32-S3 8 MB is listed for MM @wled-install

You can get it from here https://github.com/MoonModules/WLED/actions

> This reminds me of an old warning that was put into the WLED knoledge base: ![image](https://private-user-images.githubusercontent.com/91616163/385380033-09851236-64fa-46f2-b0b8-547a6535fbb4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0Mzc5MTUsIm5iZiI6MTczMTQzNzYxNSwicGF0aCI6Ii85MTYxNjE2My8zODUzODAwMzMtMDk4NTEyMzYtNjRmYS00NmYyLWIwYjgtNTQ3YTY1MzVmYmI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEyVDE4NTMzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkMjNlMzViZDVmNTI2MTNkODZhNTUyMzc0MTUxZmQ0YmFkZWVkNjUxOTQ4YWVjMTNjY2FhNDBmZThlNDNiNWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VxiKGzkD9NweJGLUaZZl7hYfPjP1SJyqwVzLDm8J1jM)
> 
> Please consider using [WLED ESP Flasher](https://kno.wled.ge/basics/install-wled-flasher), or the web installer or esptool.

I flashed after this error by esptool, but why was it working fine a month ago? And the firmware wasn't broken after uploading just the bootlooader didn't need flashing again

Canditi890, 2024-11-12:
I also have a small question about the two 128x64 panels, one works by setting up two chains because it has a 1/32 scan and can it manage to run the other one through the other HUB75 port? or is it still a problem?
I can't even create 256x64 by display errors, it either splits in half width or clones the same thing on two

dosipod, 2024-11-12:
That note is not meant to you but to  the owner of the installer site to include  HD-WF2  bin.
I do not recall how did I flash my HD-WF2 but I think I used the wled flasher linked to you so try that and call it a day .
As to the size of the panels I only used one 64x64  with  HD-WF2 but might be that is better for you to check discord as
I did not test

Canditi890, 2024-11-12:
bin will probably be added sometime from what I remember this board is still in the testing phase

softhack007, 2024-11-12:
> I also have a small question about the two 128x64 panels, one works by setting up two chains because it has a 1/32 scan and can it manage to run the other one through the other HUB75 port? or is it still a problem?

Unfortunately no, and it will be a problem for some time.
* you can't use the second HUB75 port, because the driver we use (the one from mrFaptastic) cannot handle more than one port.
* 128x64 is quite a lot of pixels for poor ESP32-S3, especially when you don't have PSRAM. So practically speaking, going beyond 128x64 will not make you happy because the esp will be very unstable.

dosipod, 2024-11-13:
I only have two 64x64 to test with but I thought the OP plan to test 256x64 might be a bit unstable but to be honest might be things could be somewhat usable even if not very stable but others would just need to spare the time to test and report back