MoonModules/WLED-MM #190: (feature request) translate effect input "bins" to corresponding frequencies

MaxMSchneider, 2024-11-20:
**Is your feature request related to a problem? Please describe.**
I have been working with the Aircookie WLED SR (archived) so far, putting up 150ft of strips on different music events. In order to precisely set up certain sound reactive effects, such as the Freqwave and Freqmatrix, it turned out to be super finnicky to get the high and low bin set up properly in order to dial in the effect as desired (depending on the music that is being played).

**Describe the solution you'd like**
Instead of just having numbers for the high and low bin, it would be nice to translate them over to the actual frequencies (Hz) they are referring to, at least in the UI. This way the setup of the effects would be much more straight forward. This would also be handy for effects that have a bin select, such as Puddlepeak for example.

I am aware that this calculation would require the sample rate of the input signal, maybe there's a way to determine the rate from the input (in my case an I2C ADC) automatically? Even the option to manually enter the sample rate for calculation in the Usermod Settings would suffice.

An alternative would be to have some sort of a cheat sheet in the SR wiki. That would be a start as well.

Thank you guys!

ewowi, 2024-11-20:
Small correction, it was https://github.com/atuline/WLED

Not aircoookie 🙂

MaxMSchneider, 2024-11-20:
@ewowi You're right, my bad!

troyhacks, 2024-12-08:
The future of AudioReactive will likely be migrated to use the ESP-DSP code as that's optimized for the fastest FFT calculations on the ESP32 family. The ESP32-S3 and ESP32-P4 both have FFT "on the chip" they are extremely fast and require very little CPU.

ESP-DSP also gives us easy access to filters, allowing us to add High-Pass (which filters out low end frequencies) and Low-Pass filters (which filter out high end frequencies) to the code that are very low on CPU use. Ideally a clean FFT should have these filters to cut frequencies that it can't understand.

MaxMSchneider, 2025-01-25:
Sorry I haven't responded so far @troyhacks! This really helped me understanding! Looking forward to what's coming in future released.

I'm planning on maybe even using WLED in "professional" setups, speaking of light art sculptures being set up on music festivals! 🎉