MoonModules/WLED-MM #200: Pre built binary for Huidu HD-WF2 for Hub 75 support

vitauser1, 2024-12-03:
Please release a pre build binary file specifically for Moon modules WLED running on Huidu HD-WF2 for HUB 75 support.

https://mm.kno.wled.ge/2D/HUB75/ shows should work with this binary "esp32S3_4MB_S" but could not see this type on releases page, only one for 8MB?

Also think it needs to redefine pins as per this link below as different to the standard pin outs
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/433.
"If you are using any other config, you currently need to edit wled00/bus_manager.cpp to add a new elif block and define to your build - it is not possible to set the HUB75 pin config in LED preferences at the moment"

Please if anyone can help as this is beyond my expertise.  alternatively a few pointer on how to do this and I can try!

dosipod, 2024-12-04:
You can look at platformio.ini here  
https://github.com/MoonModules/WLED/blob/mdev/platformio.ini

Any build with   ${common_mm.HUB75_build_flags} will have HUB75 support.I have used  esp32S3_4MB_S env and it works fine on HD-WF2, please retest and let us know if we are to double check as I am using a modified build at the moment and have only one board to test with .

[image]

A lot of testing  with HUB75 and  HD-WF2  also here https://github.com/MoonModules/WLED/issues/172  which could be good to look at

netmindz, 2024-12-04:
The bin listed in the documentation has the right pins already set. 

What happens when you try and upload the recommended bin?

vitauser1, 2024-12-05:
> The bin listed in the documentation has the right pins already set.
> 
> What happens when you try and upload the recommended bin?

Hi, thanks for replying.  I have successfully loaded WLEDMM_0.14.1-b32.40_esp32S3_4MB_PSRAM_M.bin onto the HD-WF2 board (but this does not have Hub 75 support).   When I try to load this:
https://github.com/MoonModules/WLED/releases/download/v0.14.1-beta.32/WLEDMM_0.14.1-b32.40_esp32S3_4MB_PSRAM_S.bin 
using the 'Manual OTA Update' utility in WLED after about 20 seconds it says 'Update failed! Please check your file and retry!'
HD-WF2 board is _v7.2.0-2
Any help would be appreciated.

netmindz, 2024-12-05:
Why are you using the PSRAM build not the one listed in the docs

https://github.com/MoonModules/WLED/actions/runs/12168485995/artifacts/2275990728

vitauser1, 2024-12-05:
> Why are you using the PSRAM build not the one listed in the docs
> 
> https://github.com/MoonModules/WLED/actions/runs/12168485995/artifacts/2275990728

Thanks - I didn't realise I was using the wrong one - uploaded version above and starting to play with it now - got colours and 2D working I think ! -   thanks so much for your help really appreciate it.

vitauser1, 2024-12-05:
I successfully uploaded the firmware to my HD-WF2 however the colour order seems to be wrong (Red is blue and blue is red).... any ideas on how to fix - WLED does not seems to have a colour order setting for HUB75 led types?

netmindz, 2024-12-06:
Sadly some panels are produced that do not follow the proper pinout on their connectors. Currently your only option is to edit the pin config in bus_manager.cpp

I am in the process of adding support for non standard colour ordering but that is a work in progress. Which colours are wrong on your panel?

vitauser1, 2024-12-06:
> 
> I am in the process of adding support for non standard colour ordering but that is a work in progress. Which colours are wrong on your panel?

…..Red is blue and blue is red.   Green is correctly green.

Thanks, I will try to compile a version with different pins, but I am learning how to do this…I tried yesterday but could not quite follow the guides and haven’t managed to compile anything yet!…to be honest visual studio and platformio just confuses me!  I liked the Arduino ide!

closed 2024-12-06 by vitauser1