MoonModules/WLED-MM #205: Problem with HUB75 Adafruit Matrix Portal S3

letmoc, 2024-12-17:
### What happened?

Hello, I received two Adafruit Matrix Portal S3. I then flashed the current WLED-MM binary. Adafruit MatrixPortal -S3, with HUB75 support: WLEDMM_0.14.1-b32.40_adafruit_matrixportal_esp32s3.bin.

Unfortunately, the 64x64 matrix is very delayed, and the colors don’t really match either. 

### To Reproduce Bug

Always occurs. Adafruit Matrix Portal S3

### Expected Behavior

Does anyone have an idea what could be the issue?

### Install Method

From MoonModules Release Page

### What version/release of MM WLED?

WLEDMM_0.14.1-b32.40_adafruit_matrixportal_esp32s3.bin

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

letmoc, 2024-12-17:
Video:
https://wled.discourse.group/t/problem-with-hub75-adafruit-matrix-portal-s3/13630

softhack007, 2024-12-18:
hi @letmoc, did you try the latest build? 
Its not released yet, but already very stable so you should give it a try https://github.com/MoonModules/WLED/actions/runs/12320735208

The color differences might be due to changes on the panel manufacturer side - its quite normal that panels purchased at different times (even from same supplier) will not match colors.

letmoc, 2024-12-18:
Hi @softhack007 
Thank you for your feedback. I will test the firmware later. The displays are both the same; when you swap them, the color differences also shift. I will provide feedback later.

letmoc, 2024-12-18:
@softhack007

Update failed!
Please check your file and retry!

An update fails. When I flash the .bin firmware using esp-flasher, it goes through, but afterwards the WLED-AP is not displayed. The board is not working. Firmware: WLEDMM_14.5.0-beta_adafruit_matrixportal_esp32s3.bin

softhack007, 2024-12-18:
> The board is not working. Firmware: WLEDMM_14.5.0-beta_adafruit_matrixportal_esp32s3.bin

@netmindz can you test the file? I don't have a matrixportal board.

softhack007, 2024-12-18:
> using esp-flasher

@letmoc is this "esp home flasher"? I yes, then please use a different tool. ESP home flasher creates broken installs.
[WLED ESP Flasher](https://kno.wled.ge/basics/install-wled-flasher) should work better.

softhack007, 2024-12-18:
Another option for flashing:
Step 1. Use this web-based installer to flash any S3 (8MB with qspi psram) firmware: https://wled-install.github.io/
Step 2. OTA update to the adafruit_matrixportal firmware

letmoc, 2024-12-19:
@softhack007

MoonMod WLED Software Update
WLEDMM_0.14.1-b32.40 matrixportal_esp32s3.bin
(ESP32-S3 build 2406240)

MoonMod WLED Software Update
Updating...
Please do not close or refresh the page :)

Update failed!
Please check your file and retry!

++++++++++++++++++++++++++++++++++++++

The OTA update fails. I used the WLED ESP Flasher, and after flashing, the board does not start, and no WLED-AP is displayed.

letmoc, 2024-12-19:
Using 'COM7' as serial port.
Connecting...
Detecting chip type... ESP32-S3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-S3
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: NO
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: NO
 - MAC Address: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 8MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.9s
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001a9fff...
Flash params set to 0x023f
Compressed 13936 bytes to 9695...
Writing at 0x00000000... (100 %)
Wrote 13936 bytes (9695 compressed) at 0x00000000 in 0.2 seconds (effective 732.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 894.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1284.2 kbit/s)...
Hash of data verified.
Compressed 1675872 bytes to 1136744...
Writing at 0x00010000... (1 %)
Writing at 0x00019165... (2 %)
Writing at 0x0001e1b5... (4 %)
Writing at 0x00022235... (5 %)
Writing at 0x0002624d... (7 %)
Writing at 0x0002a263... (8 %)
Writing at 0x0002e25c... (10 %)
Writing at 0x000321d8... (11 %)
Writing at 0x0003b35e... (12 %)
Writing at 0x00046b80... (14 %)
Writing at 0x0004f045... (15 %)
Writing at 0x00060659... (17 %)
Writing at 0x00065a12... (18 %)
Writing at 0x0006aab9... (20 %)
Writing at 0x0006fd5c... (21 %)
Writing at 0x00074cd8... (22 %)
Writing at 0x0007a05f... (24 %)
Writing at 0x0007fd1f... (25 %)
Writing at 0x000853cc... (27 %)
Writing at 0x0008a655... (28 %)
Writing at 0x0009076d... (30 %)
Writing at 0x00096298... (31 %)
Writing at 0x0009c224... (32 %)
Writing at 0x000a2782... (34 %)
Writing at 0x000a8059... (35 %)
Writing at 0x000ad330... (37 %)
Writing at 0x000b2dd9... (38 %)
Writing at 0x000b8a4a... (40 %)
Writing at 0x000bdd4b... (41 %)
Writing at 0x000c333f... (42 %)
Writing at 0x000c9639... (44 %)
Writing at 0x000cf638... (45 %)
Writing at 0x000d51b7... (47 %)
Writing at 0x000db876... (48 %)
Writing at 0x000e1811... (50 %)
Writing at 0x000e71e9... (51 %)
Writing at 0x000ed776... (52 %)
Writing at 0x000f36fb... (54 %)
Writing at 0x000f907e... (55 %)
Writing at 0x000fec0b... (57 %)
Writing at 0x00104714... (58 %)
Writing at 0x00109d71... (60 %)
Writing at 0x0010fbc1... (61 %)
Writing at 0x00114d9d... (62 %)
Writing at 0x00119b91... (64 %)
Writing at 0x0011eb03... (65 %)
Writing at 0x00123ce6... (67 %)
Writing at 0x00128be3... (68 %)
Writing at 0x0012db6e... (70 %)
Writing at 0x00132d8d... (71 %)
Writing at 0x00137a36... (72 %)
Writing at 0x0013cb16... (74 %)
Writing at 0x0014224f... (75 %)
Writing at 0x001479f8... (77 %)
Writing at 0x0014c726... (78 %)
Writing at 0x0015185e... (80 %)
Writing at 0x001567aa... (81 %)
Writing at 0x0015bc7e... (82 %)
Writing at 0x00161068... (84 %)
Writing at 0x00166a5b... (85 %)
Writing at 0x0016c34b... (87 %)
Writing at 0x001718cb... (88 %)
Writing at 0x00176d9e... (90 %)
Writing at 0x0017f186... (91 %)
Writing at 0x001877c5... (92 %)
Writing at 0x0018d7ab... (94 %)
Writing at 0x001963c0... (95 %)
Writing at 0x0019b816... (97 %)
Writing at 0x001a1148... (98 %)
Writing at 0x001a6db0... (100 %)
Wrote 1675872 bytes (1136744 compressed) at 0x00010000 in 10.7 seconds (effective 1252.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

letmoc, 2024-12-19:
[image]

softhack007, 2024-12-19:
Ok, I did some research. It seems that adafruit wants it "their way", so its best to use their tools and use their stock bootloader.

1. follow adafruit instructions to install the adafruit "tinyUF2" factory-reset-and-bootloader firmware: https://learn.adafruit.com/adafruit-matrixportal-s3/factory-reset#factory-reset-and-bootloader-repair-3107941
    (use the adafruit web flasher)

2. unzip the attached files, and flash them with the adafruit web flasher https://adafruit.github.io/Adafruit_WebSerial_ESPTool
   * offset `0x8000` partitions_8MB_tinyuf2.bin
   * offset `0x10000` WLEDMM_14.5.0-beta_matrixportal_esp32s3_tinyuf2.bin
   * hit "Program"

3. restart your matrixportal board. This should give you a working WLED-MM installation.

"should" because I don't have the matrixportal board to test.

[matrixportal_tinyUF2.zip](https://github.com/user-attachments/files/18197561/matrixportal_tinyUF2.zip)

letmoc, 2024-12-19:
You are the best! 
It works now. 
The matrix behaves identically now, and the colors are as they should be.

softhack007, 2024-12-19:
@letmoc thanks for reporting the problem - it looks like we should update our HUB75 documentation for matrixportal.

letmoc, 2024-12-19:
@softhack007 
Thank you for your quick help. 
I also flashed the second board in this way. It worked immediately. 
I'm curious if a future OTA update would work.

softhack007, 2024-12-19:
> I'm curious if a future OTA update would work.

Yes it should work. OTA can only replace the firmware but the bootloader and partition file that you have now will not be replaced. The OTA partition is about 2MB now, this should be good for longer time.

netmindz, 2024-12-20:
Sorry for the slow reply. I can't find where I've put my matrix board at the moment. 

I have only flashed using platformio and then using the OTA I think.

It's a long time ago since I first got the board so it's possible I'm forgetting a step.

Using a regular S3 build from web installer then OTA I would expect to work.

I possibly needed to press one or more buttons actually, but I don't recall if that was for the first flash or recovery from a soft brick while doing the initial development work

fonix232, 2025-01-09:
I've had this exact same issue on the MPS3, and practically none of the listed solutions worked:

- tried using the official WLED flasher (https://install.wled.me/) - wouldn't boot
- tried using the unofficial WLED web flasher (https://wled-install.github.io/) - wouldn't boot
- tried using the ESPHome web installer (https://web.esphome.io/) - wouldn't boot
- tried using the desktop WLED flasher (https://mm.kno.wled.ge/basics/install-wled-flasher/) - wouldn't boot
- tried ESPTool.js (https://espressif.github.io/esptool-js/) - wouldn't boot

On each method, I tried both directly flashing the WLEDMM MatrixPortal S3 build, as well as a generic WLED ESP32-S3 build, to no avail.

What DID work was using the basis of the ESPHome flasher - [ESP Web Tools](https://esphome.github.io/esp-web-tools/). This bad boy successfully flashed the stock S3 build, from which I managed to OTA update to the MatrixPortal-specific build.

I've also managed to repeat this process without any issues, while the other tools, once the flash was wiped (and many of them completely wipes the flash at the beginning of the process), would result in a non-booting unit.

letmoc, 2025-01-09:
Hello, I have the following problem with the board: it suddenly reboots. Attached is a screenshot. I am just running Akemi for testing. Then, after a few hours, sometimes even over 24 hours, it reboots.

[image]

softhack007, 2025-01-09:
> @softhack007
> Hello, I have the following problem with the board: it suddenly reboots.

@letmoc I don't have the adafruit board, but i could did out a similar board with qspi flash and let it run for a day to check if i can reproduce the crash - "sw error" is not very specific, it could be anything - out-of-memory, bug in  the code, bug in the HUB75 driver, etc.

* How did you configure the led panel (screenshot from LEDs settings, please) ?
* does it also happen when running the board alone, without the panel connected ? 
   (you can simply disconnect the HUB75 cable, as the software does not care if anything is connected to the HUB75 port or not)

letmoc, 2025-01-10:
@softhack007 
I have two boards, and I updated one matrix portal to version 14.5.1-dev build 2501030. It has been running for exactly 24 hours. For testing, I have two different 64x64 displays with different driver ICs. Both behave the same. Attached are the screenshots.

[image]
[image]

softhack007, 2025-01-10:
> I have two different 64x64 displays with different driver ICs. Both behave the same

Sorry if I repeat myself - please disconnect the panel and let one board run "headless". It will help me to understand if the issue is reproducible when minimizing electrical "disturbances" created by the panels.

letmoc, 2025-01-11:
@fonix232 
> What DID work was using the basis of the ESPHome flasher - [ESP Web Tools](https://esphome.github.io/esp-web-tools/). This bad boy successfully flashed the stock S3 build, from which I managed to OTA update to the MatrixPortal-specific build.

I can confirm it works perfectly like this, dear WLED friend. :-)

softhack007, 2025-01-11:
> I can confirm it works perfectly like this, dear WLED friend. :-)

If you're happy with it, that's OK; ESP Web tools is also used by the official WLED web installer. 

### Just keep in mind you have been warned about *ESPhome Flasher*.

It might be that a few months after using this tool, you'll  see are crazy errors (e.g. cannot save preset or OTA update a new release errors out).

I'll simply dismiss any problem report after you used ESPHome Flasher. Same about the other wild ideas of first installing Tasmota, then OTA update with a WLED binary.

softhack007, 2025-01-11:
> I can confirm it works perfectly like this, dear WLED friend. :-)

@fonix232  @letmoc  If you're happy with it, that's OK; ESP Web tools is also used by the official WLED web installer. 

### Just keep in mind you have been warned about *ESPhome Flasher*.

It might be that a few months after using this tool, you'll  see are crazy errors (e.g. cannot save preset or OTA update a new release errors out).

I'll simply dismiss any problem report after you used ESPHome Flasher. Same about the other wild ideas of first installing Tasmota, then OTA update with a WLED binary.

softhack007, 2025-01-11:
If you have a mateixportal board, I strongly recommend that you use the bootloader and tools from adafruit. Procedure is described [above](https://github.com/MoonModules/WLED/issues/205#issuecomment-2553617390). 

After initially flashing the adafruit firmware, then uploading  partitions.bin and WLEDMM using adafruit web tools, you can safely use OTA for installing any WLED bin by OTA.

letmoc, 2025-01-14:
@softhack007 

The second module also rebooted. Just for your information.

netmindz, 2025-01-25:
Can we open a fresh issue please to track reboots and then we can close this installation one as resolved

letmoc, 2025-02-01:
@netmindz @softhack007 
I have created a new topic; the discussion is now continued under https://github.com/MoonModules/WLED-MM/issues/222

netmindz, 2025-02-03:
Are we now good to close this issue?

Sireone, 2025-08-16:
After following the reset step from @softhack007 above & flashing both files, I still don't see any WLED-AP.  Thoughts? 

[image]

softhack007, 2025-11-22:
@Sireone did you also perform this step

> follow adafruit instructions to install and run the factory-reset-and-bootloader firmware: https://learn.adafruit.com/adafruit-matrixportal-s3/factory-reset#factory-reset-and-bootloader-repair-3107941, and restart your board afterwards.
(use the adafruit web flasher aka "WebSerial ESPTool Method")

before running the adafruit web flasher? Its necessary to restore the adafruit special bootloader. If you don't have the adafruit bootloader, nothing will work.

Also keep your computer connected via USB and watch messages on serial monitor when starting WLED-MM on the matrixportal - without this we can't provide help.